Atmosphere-NX/Atmosphere #965: Panic Error when i run hbmenu after starting a game or app

johner1, 2020-05-26:
**Error**
Panic Occurred while running Atmosphere.

Title ID: 010041544D530000
Error Desc: std::abort() (0xFFE)]

**Error report:**
[report_458f248d3.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4679485/report_458f248d3.zip)

I updated my fw 9.1 to 10.0.2 & ams to 0.12 from 0.92 
Everything was working well but when i try to start hbmenu after opening any game/tinfoil. It gives that sudden error.

### Methods applied:
1. I did the fix bit used ----all good.
2. I deleted sys-clk sys-con they gave new error so rerolled.
3. I use Fat32 yet i downgraded to 10.0.1 then again upped to 10.0.2 on fat32 then i moved my important Ntendo file to root.Started through hekate to primary fusee --found same error.
4. Tried the Ram issue--- deleted applet programs 1/1 to find fix which didn't change a thing.

**Fw & Ams**

10.0.2 & 0.12

What should i do ??

fennectech, 2020-05-26:
I see this panic with 4 custom sysmodules   Seems to be memory related   (sys-con sys-ftpd-light sys-clk and tesla)

johner1, 2020-05-26:
i checked out the issue with much more deliberation & found that it was problem with the ams package i replaced it with the latest one from git which was lastly from sdsetup. Btw,thanks.

PrimetimeFluff, 2020-05-27:
I'm having the same issue. What did you do to fix it?
I've updated AMS from github.

johner1, 2020-05-28:
just replaced with the ams from git...guess there were some diff files from sdsetup with those ams & sept folder.And this fixed the issue.Tried many methods you can check up.

SciresM, 2020-05-28:
Not an atmosphere issue.

Sysmodules are exhausting the FS session limit (it appears most custom sysmodules outside atmosphere are not configuring themselves to only use one FS session instead of three, and are counting triple against the limit).

This is a fixed, unpatchable system limitation -- the developers of the sysmodules you use will need to release better behaved builds which correctly respect the limitations.

I'm reaching out to some sysmodule developers (e.g. https://github.com/exelix11/SysDVR/issues/58) to get them to do this where I can.

closed 2020-05-28 by SciresM