Atmosphere-NX/Atmosphere #967: Fatal error occured when running atmosphere

Ex0hm, 2020-05-26:
## Bug Report

### What's the issue you encountered?

I updated from 9.2exfat to 10.0.2exfat with Choi after rebooting when i tried to play a game it prompted me if i want to update or start the software instead of starting I accidentally pressed update and now I'm on 10.0.3 and it asks me to update my sdxc to be able to use exfat but when i try to update it says firmware up to date so i rebooted to payload and this is the error im getting

### Did you make any changes related to Atmosphère itself?
 No

### How can the issue be reproduced?

Updating to latest software via NSP?

### Crash Report

A fatal error has occurred when running atmosphere
Title ID: 010041544d530000
Error desc: std : : abort() called (0xffe)
Report saved to /atmosphere/fatal_errors/report_0000000005ec2477.bin

https://github.com/Ex0hm/AMS-Error_report/blob/master/report_0000000005ec2477.bin

### System Firmware Version

10.0.3

### Environment?

Bootloader: Fuseè
AMS 0.12.0

- Homebrew software installed: 
Edizon
Awoo installer
Reboot to payload

mikookoi, 2020-05-27:
man i have the same error i don know how fix that

Ex0hm, 2020-05-27:
> man i have the same error i don know how fix that

I found out that it wasn't an error from atmosphere but from the firmware it updated to instead of having the exfat driver it installed the non exfat version of the firmware, what I'm doing right now is using my 32gb SDHC card to reupdate my current 10.0.3 to 10.0.3 exfat by using choi but Incant seem to find any 10.0.3 Firmware dump.

Ex0hm, 2020-05-27:
> I found out that it wasn't an error from atmosphere but from the firmware it updated to instead of having the exfat driver it installed the non exfat version of the firmware, what I'm doing right [...]

Tried it but i can't update my current 10.0.3 to 10.0.3exfat, so I really don't know how to solve this and I don't know if there is a way to just install the exfat driver manually since nintendo just says your firmware is already up to date.

SciresM, 2020-05-28:
@Ex0hm Format your SD card as FAT32, or overwrite package2 with exFAT firm manually. Either way, not an atmosphere issue, heh...

closed 2020-05-28 by SciresM