Atmosphere-NX/Atmosphere #1005: Boot is broken

Anti-Apple4life, 2020-05-29:
## Bug Report

Whenever I boot Atmosphere, it shows the black-and-white logo, but does not boot,

### What's the issue you encountered?
Whenever I boot Atmosphere, it shows the black-and-white logo, but does not boot,

### How can the issue be reproduced?
I don't know

### Crash Report

No report made

### System Firmware Version

1.2.0
### Environment?
Hekate and Fusee
### Additional context?

SciresM, 2020-05-29:
Closing, please reopen using the issue templates properly. "Doesn't boot" is useless; say what you observe fully when reopening.

1.2.0 is not a real firmware version.

closed 2020-05-29 by SciresM

Anti-Apple4life, 2020-05-29:
I had a typo  and I tried to put 9.2.0 and The boot sequence goes until the black-and-white ams logo, but then freezes.

cheesycod, 2020-06-03:
If you're still having this problem, exactly what version of atmosphere are you using?

SciresM, 2020-06-03:
This was a troll issue, @cheesycod. Please do not give it more attention.

SciresM, 2020-06-03:
Oh, never mind, this wasn't the 10.1.0 guy. Fwiw issue after the b/w ams logo is "typically" exfat or archive bits -- some failure to read contents on the SD.

I apologize, there was a nearly identical troll issue the other day.