Atmosphere-NX/Atmosphere #1006: Random panic errors keep occuring.

SonicvMario, 2020-05-29:
## Bug Report

### What's the issue you encountered?

When I'm playing my Switch I keep getting panic errors with the title_id 010041544D530000 and the error description std::abort<> <0xFFE> while running Atmosphere 11.0 . This started happening on Wednesday and prior to that I hadn't played the Switch for about a week though it was working perfectly fine before that. I tried the archive bit fixer in Hekate v5.1.4 before updating atmosphere to 12.0 and hekate to 5.2.1 neither of which helped to solve the issue. The crashes have occurred only while in game or on boot, never on the home menu although this is probably just luck due to the randommess of it.  Below is a picture of the crash screen
[image]

### How can the issue be reproduced?

I can't find a consistent way of reproducing the issue due to its randomness. Sometimes I can leave my Switch running for a few hours before it crashes, sometimes it happens within minutes of a reboot but it always happens eventually. 

### Crash Report
A crash report doesn't seem to be getting created when the crash occurs. The crash screen says that one is being made in atmosphere/fatal_errors but the file is never created. 

### System Firmware Version
10.0.0

### Environment?
Hekate 5.1.4 + Atmosphere 11.0 (Aquired using Kosmos) 
Hekate 5.2.1 + Atmosphere 12.0 (Aquired directly from github) 

Sysmodules: Tesla, Sys-clk & Sys-con

Kips: ams_mitm.kip, loader.kip (think these came with a kosmos installation but I'm unsure)

### Additional context?
Due to the randomness of the crashes, the fact that the archive bit fix didn't do anything and the lack of error reports being made I'm thinking that it may be an sd card issue. I thought it would probably be better to submit a report first though rather than going straight to grabbing a new sd card.

nwhiteman42, 2020-05-31:
What is your SD card formated as?  FAT32 or exFat?  Have you tried running without your sysmodules?

SonicvMario, 2020-05-31:
It's formatted as Fat32. Doing a bit more testing though the sd card
seems to be working fine on all of my other devices and the random crashing
also happens using other cfws. I'm thinking it's an issue with the sd card
reader in my Switch so I'll probably just close the issue.

cheesycod, 2020-06-03:
did you try formatting your sd card and installing atmosphere again

SonicvMario, 2020-06-03:
Yeah gave it a go a few times but it didn't seem to work. I'm currently waiting on a delivery of a tri-wing screwdriver so I can re-seat the sd card reader and if that doesn't work I'll replace it.

cheesycod, 2020-06-03:
is hekate able to read the SD card properly?

SonicvMario, 2020-06-03:
It can most of the time but occasionally it will say something along the lines of the sd had been initiated in 1 bit mode and sometimes it will hang when loading boot configs.

cheesycod, 2020-06-03:
I think you might need to replace the sd card then. Also, do you have a known working nand backup

SonicvMario, 2020-06-03:
I doubt it's an sd card problem tbh, like I said earlier it works fine on everything else just not my Switch. It's most likely the card reader in the Switch which I'll try re-seating when my tri-wing screwdriver arrives, if that doesn't work I've got a replacement sd reader on the way as well.

dazzaXx, 2020-06-04:
I'm suddenly getting the same issue but with "010000000000002b" running 0.12.0 on 10.0,3.

SonicvMario, 2020-06-08:
After opening it up it looks like the sd reader connector has a missing pin. 
[image]

Apparently that's fixable by soddering a wire directly from the pins to the sd reader but I don't have the skills for that so my Switch is now stuck in 1 bit mode for the sd card. Still usable but incredibly slow to load anything. On the plus side I haven't had any panic errors occur after leaving my Switch running for a few hours now so "touchwood" they were probably being caused by the unexpected switches between 4bit and 1 bit mode.

closed 2020-06-28 by SciresM

dazzaXx, 2020-06-28:
I commented on here a while ago, would just like to say that my issue was being caused by a failing sd card, ordered a new one and all is working again.