Atmosphere-NX/Atmosphere #1010: Opening webapplet from applet mode freezes and eventually crashes

HookedBehemoth, 2020-05-31:
## Bug Report

### What's the issue you encountered?

When opening the web applet from applet override it freezes for about 30 seconds and eventually crashes.

I would expect that to work as official software album and qlaunch can use loginShare.

Yellows8 sent me here:
> Webapplet is failing to allocate enough heap. Same thing with `applet_heap_size = u64!0x4000000`
> 
> This isn't really libnx related (AMS repo would be more relevant(?)), so closing.

Tested various ram reserve sizes.

At 0x16000000 it opens but fatals on exit. No crash report is generated. [Fatal report](https://github.com/switchbrew/libnx/files/4705665/01590796465_010000000000100a.log)
Anything I tested below that froze. Above that [hbmenu wasn't useable.](https://user-images.githubusercontent.com/22580720/83329358-521be300-a289-11ea-9ff6-f152dd2a9b15.jpg)

### How can the issue be reproduced?

Use the [web applet example.](https://github.com/switchbrew/switch-examples/tree/master/applet/libapplets/web)

### Crash Report

https://github.com/switchbrew/libnx/files/4705661/01590798361_010000000000100a.log

### System Firmware Version

10.0.x/4.0.1</br>

### Environment?

- Tested on official atmopshère build with hekate.
- Various reserved ram sizes tested. [See original issue thread.](https://github.com/switchbrew/libnx/issues/477)
- Tested w/o any custom sysmodules

SciresM, 2020-05-31:
Okay. So, what is the actual issue and what are you hoping to see solved?

SciresM, 2020-05-31:
I am mostly unsure what you would like me to do.

HookedBehemoth, 2020-05-31:
actual issue: crashes
what are you hoping to see solved: it crashing

I'd like to be able to use the web applets from album.

SciresM, 2020-05-31:
Do you have a memory range at which it works?

HookedBehemoth, 2020-05-31:
reserving 0x16000000 opens it but closing it fatals without a crash report.

SciresM, 2020-05-31:
I do not feel reserving that much is worthwhile.

Applet mode Homebrew operates under memory constraints. If you can find a memory range where stuff works, great -- it'll be considered.

If not, I honestly do not consider it important that applet mode homebrew be able to launch webapplet. 

Applet mode homebrew is already intended to not fully support all homebrew features due to its memory constraints. You can always adjust your own reservation value to suit your use case.

Keep me posted, I guess.

closed 2020-05-31 by SciresM

HookedBehemoth, 2020-05-31:
I saw this as an issue as official software (album) still uses roughly the same amount of RAM as hbl+atmosphere reserves but is able to use the web applets.

I didn't want to suggest actually reserving that much memory. It was just a value that yls suggested that actually opened it.

SciresM, 2020-05-31:
HBL uses *substantially* more memory than qlaunch/applet, even taking reservations into account.

HookedBehemoth, 2020-05-31:
I checked [with this](https://github.com/masagrator/Status-Monitor-Overlay) which uses [svcGetSystemInfo](https://github.com/masagrator/Status-Monitor-Overlay/blob/master/source/main.cpp#L229) for getting used vs total ram usage.

Tells me that on default settings with hbl applet RAM usage is ~345MB and album ~354MB out of the available 472MB.

SciresM, 2020-05-31:
I think the get system info call only accounts for normal memory, and not other memory states.

The real question imo is resource usage at web applet launch time/web applet heap spawn time.

What are memory statistics in that context?

HookedBehemoth, 2020-05-31:
Ok it drops down do the amount applet occupies without album running.

So ofw just closes it completly. Oh well...

Thanks for the help.

SciresM, 2020-05-31:
Yeah, OFW closing applet isn't really a choice we have for HBL lol :)

HookedBehemoth, 2020-05-31:
If only libapplet homebrew was supported...