Atmosphere-NX/Atmosphere #1023: Allow other permissions for memory mapped using MapProcessMemory then R/W

RSDuck, 2020-06-01:
## Feature Request

### What feature are you suggesting?
#### Overview:

Allow the permissions of memory mapped using MapProcessMemory to be changed for each page individually, instead of having everything always set to R/W.

#### Smaller Details: 

N/A

### What component do you feel this would best fit within?

Unknown

### Why would this feature be useful?

A common optimisation for emulators is called fast mem. The idea is too speed up the emulation of memory accesses by mapping memory into a region of virtual memory so that the layout matches the one found on the emulated system (this includes e.g. memory mirrors). This allows to radically simplify the code generated by the JIT compiler for a load/store. On Horizon this is usually done using svcMapProcessCodeMemory to create a readonly section which then can be mapped into the desired places using MapProcessMemory.

While accesses to I/O registers can be easily catched using the exception handler, since they fall into unmapped address space, writes which override code (or for example textures or framebuffers) needs to be detected. For this it's necessary be eable to set the mapped memory permission to readonly.

Alternatively to this (though that would deviate more from standard behaviour) would be to allow for some other more straightforward way to mirror heap memory (similar to svcMapMemory, but without making the previous region unusable.

SciresM, 2020-06-01:
I mean, no I will not add this, but why does it not work with the existing primitives? What error do you get from setmemoryperm/setprocessmemoryperm

RSDuck, 2020-06-01:
the error is d401 or invalid state/permissions. It can be reproduced using [this code snippet](https://gist.github.com/RSDuck/4d65dd3d980aceb52e5fb5a843fceb63).

I'm fully aware that this is not what these system calls are meant for and that this is in no way the proper way of doing things. Though at the same time I don't know any other way this could be accomplished/this is the nearest you get at the moment.

On other OSes this is usually done using memory mapped files (backed by memory).

SciresM, 2020-06-01:
@RSDuck I don't understand. Why not just put the code memory with correct permissions at the desired location in the first place? Why put the RW- non-executable view at the location you actually intend to execute?

RSDuck, 2020-06-01:
In the snippet I sent, the "code mem" (with permissions rx) is never meant to be executed from. It's just an immediate step to be eable to remap the memory (which is used as the memory of the emulated system) at any place in the address space (sometimes multiple times). This is part of what makes this workaround so emberassing.

SciresM, 2020-06-01:
Yeah but like instead of doing

```
    svcMapProcessCodeMemory(curProcess, (u64)code_address, (u64)backing_memory, 0x1000);
    svcMapProcessMemory(desired_address, curProcess, (u64)code_address, 0x1000);
```

Why not just do

```
svcMapProcessCodeMemory(curProcess, desired_address, (u64)backing_memory, 0x1000);
```
?

RSDuck, 2020-06-01:
the problem with this, is that you can't do mirrors with this, as it fails after the second call of svcMapProcessCodeMemory

SciresM, 2020-06-01:
Still not sure that I see the problem, @RSDuck.

Either your memory is RW- or it is R-?.

In the first case, you can create as many mappings as you want.

In the second, you can only create one, yes. 

One option would be to maintain multiple backing stores. Since you're trapping and executing an exception handler on memory modification, you can afford to simply maintain two backing copies and write to both inside the exception handler perf-wise -- the cost of an additional u32 write to synchronize the extra backing pages compared to the cost of executing an entire exception handler is totally negligible.

RSDuck, 2020-06-01:
> Either your memory is RW- or it is R-?.

yes and no, ROM can be mapped as read only, but we're leaving this aside for a moment. With a JIT recompiler, individual pieces of guest code is recompiled to aarch64 code. These recompiled blocks are of course cached instead of being recompiled every time. But this means that this cache needs to be invalidaded once the guest code is overriden.

> One option would be to maintain multiple backing stores. Since you're trapping and executing an exception handler on memory modification, you can afford to simply maintain two backing copies and [...]

not every memory access generates an exception, preferable as few as possible. With fast mem, memory is mapped so that a memory access can be translated into nothing more than a single memory access to the guest address with an offset, to the start of the virtual memory region where the memory layout of the guest system is recreated.

Putting this all togther there are two problems:
1. memory mapped I/O needs to be redirect to calling respective handlers of the emulator. This is done by leaving the regions where they reside unmapped, so they can be taken care of in the exception handler.
2. catching guest code being overriden. Like I said above, the JIT block cache needs to stay coherent, so writes to code memory need to be tracked. One way of doing it would be to set the permissions of the fastmem mapping of the pages which contain guest code to readonly, so they can be handled by the exception handler. This is what's problematic about the whole thing.

SciresM, 2020-06-01:
What you said did not address my points at all. What I said covers your concerns, I do not see the problem under the scheme I provided.

SciresM, 2020-06-01:
Allow me to be more specific. Suppose you have some page that needs to be mapped twice, and both mappings need to be read-only.

```
u8 *backing_page1 = // ... ;
u8 *backing_page2 = // ... ;
svcMapProcessCodeMemory(curProcess, desired_address1, backing_page1, size);
svcMapProcssCodeMemory(curProcess, desired_address2, backing_page2, size);
svcSetProcessMemoryPermission(curProcess, desired_address1, size, Perm_R);
svcSetProcessMemoryPermission(curProcess, desired_address2, size, Perm_R);
```

You now have two R-- mappings at the desired addresses.

When your emulator writes to these mappings, your exception handler will execute.

Instead of only updating one mapping, update both.

How does this not solve the problem?

SciresM, 2020-06-01:
`yes and no, ROM can be mapped as read only, but we're leaving this aside for a moment`

You did not understand what I meant.

? is a stand-in for arbitrary.

`Either your mapping is RW- or R-?`` meant `Either your mapping is RW- or R-*`, which meant `Either your mapping is RW- or R-- or R-X`.

It was not a question.

closed 2020-06-01 by SciresM

RSDuck, 2020-06-01:
I think I know where my misunderstanding was. So you're suggesting that I should map each page individually, instead of whole blocks where the permissions are set per page? I assumed that wouldn't really be feasible (coming from Windows where you can't even map at page but only at 64kb granularity), though if you're saying that works I'll happily try that.

SciresM, 2020-06-01:
You can map at 4 KiB granularity on switch. 

Even if you can't, at the end of the day it's all code memory -- map the whole code memory as desired, and then set the memory permissions on the code memory. You can set the memory permissions on the mapped code memory arbitrarily with 4 KiB granularity.

RSDuck, 2020-06-01:
whoa that really worked. I assumed things would start to crash and burn if I call svcMapProcessCodeMemory 1024 times to map a 4mb region, but it succeded everytime.

Thank you for your time, I hope I haven't annoyed you too much.

SciresM, 2020-06-01:
np.

You also don't need to call svcMapProcessCodeMemory 1024 times -- you can call it once, and then call svcSetProcessMemoryPermission 1024 times. That will work too.

SciresM, 2020-06-01:
Anyway, glad that the suggestion is workable ^^