Atmosphere-NX/Atmosphere #1024: Sysmodules with specific title IDs will not run

bakugo, 2020-06-03:
## Bug Report

### What's the issue you encountered?

Sysmodules (with the boot2 flag) with specific title IDs will not run.
Examples of IDs that will not run: 0100FFFFFFFFFFFF 01004E4554484944

### How can the issue be reproduced?

* Take the [example sysmodule from switchbrew](https://github.com/switchbrew/switch-examples/tree/master/templates/sysmodule)
* Add some code to write something to the SD card
* Compile it (with the default titleid of 0100000000000123)
* Put it on the SD and boot, it will run and write the file as expected
* Change the titleid to 0100FFFFFFFFFFFF or 01004E4554484944 and recompile
* Attempt to run it again, it will not work (file will not be written), no errors

### System Firmware Version

10.0.3

### Environment?

Atmosphere 0.12.0, nothing else added except the module in question.

### Additional context?

This issue did not occur on 0.10.2 with firmware 9.1.0.

WerWolv, 2020-06-03:
Starting with 10.0.0, Nintendo moved the ACID signature check to Loader which Atmosphere replicates too. This means now however, that custom sysmodules who's programid lands in the Application region will now be signature checked. This check obviously always fails for custom sysmodules and Loader refuses to launch it.

SciresM, 2020-06-03:
Yeah, don't use application range ids.

closed 2020-06-03 by SciresM

bakugo, 2020-06-03:
Thanks for the info. Is this documented anywhere? What ranges exactly are considered applications?

friedkeenan, 2020-06-03:
The [range](https://github.com/Atmosphere-NX/Atmosphere/blob/master/libraries/libstratosphere/include/stratosphere/ncm/ncm_content_meta_id.hpp#L35-L36) is 0x0100000000010000 to 0x01FFFFFFFFFFFFFF inclusive

bakugo, 2020-06-03:
Got it, thanks.