Atmosphere-NX/Atmosphere #1025: Compilation error

rodis120, 2020-06-03:
I have a problem with compiling the atmosphere.
arm-none-eabi-gcc: fatal error: cannot execute '/opt/devkitpro/devkitARM/libexec/gcc/arm-none-eabi/10.1.0/cc1' : execv: Execv format error compilation terminated.

cheesycod, 2020-06-03:
I don't think source code compilation is supported. Also please use the template

SciresM, 2020-06-03:
Ignoring the troll; that seems like a bad devkitARM installation. Is this triggered on compiling a specific component/file?

rodis120, 2020-06-03:
I write make in command line and the problem starter while compiling exosphere

cheesycod, 2020-06-03:
thats odd, does exosphere even use devkitARM?

SciresM, 2020-06-03:
exosphere contains sc7fw, lp0fw, and rebootstub, all of which are bpmp payloads and thus use devkitARM.

With due respect @cheesycod -- the atmosphere issue comments aren't a general purpose forum. If you aren't helping to diagnose, please avoid commenting -- your initial shitpost was unhelpful.

The likely problem is just a bad devkitARM install.

cheesycod, 2020-06-03:
Oh, sorry.

rodis120, 2020-06-03:
I didn't try it jet.

rodis120, 2020-06-03:
I reinstalled devkitpro and now i have another error:
/home/ubuntu/Downloads/ROGAL-master/exosphere/src/configitem.c: In function 'configitem_get':
<command-line>: error: invalid suffix "x" on integer constant
<command-line>: note: in definition of macro 'ATMOSPHERE_RELEASE_VERSION_HASH'

HookedBehemoth, 2020-06-03:
Atmosphère's build system relies on being inside a git repository.
Downloading the source ZIP won't work.

Either clone the repo or remove the parts in code that tries to get the build hash.

closed 2020-06-03 by SciresM