Atmosphere-NX/Atmosphere #1027: Fatal error ocurred when running Atmosphére #010041544d530000

Daypayz, 2020-06-05:
Bug Report

A fatal error occurred when running Atmosphere.
 Title ID: 010041544d530000
 Error Desc: std::abort() called (0xffe)
 Report saved to /atmosphere/fatal_errors/report_0000000061b2685.bin

What's the issue you encountered?

After booting up Atmosphere, through Hekate, screen goes black and I receive the fatal error message

How can the issue be reproduced?

Boot to RCM
 Launch Hekate
 Boot Atmosphere through either fusèe-primary or emunand
[Fatal.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4735795/Fatal.zip)

I read enough of these issues to see you can tell what the issue is. Any help would be appreciated

SciresM, 2020-06-05:
Invalid FAT filesystem.

Format your SD card as FAT32 instead of exFAT.

closed 2020-06-05 by SciresM

Daypayz, 2020-06-05:
manually selected 10.0.3 and not exfat? cheers

masagrator, 2020-06-05:
You have exFAT card and installed FW update without exFAT support.
Congrats.

Daypayz, 2020-06-05:
and there are no sig patches for 10.0.3?

SciresM, 2020-06-05:
Atmosphere doesn't support piracy.