Atmosphere-NX/Atmosphere #1030: A fatal error occurred when running atmosphere

Santiav90, 2020-06-06:
## Bug Report

### What's the issue you encountered?

##Every so often after the last update my switch crashes saying "A fatal error ocurred when running atmosphere". I haven't been able to find out when this error happens since it seems random. It also says "Title ID: 010041544d530000" and "Error desc: std::abort() called (0xffe) and report saved to /atmosphere/fatal_errors/report_000000118b4aca06.bin

### How can the issue be reproduced?

##havent found a way yet

### Crash Report

##https://gist.github.com/Santiav90/088bb424fe1bd4fe51f907d14803edfb

### System Firmware Version

##lastest update from yesterday june 5th

### Environment?

##I use a fuseeloader and the latest official release

### Additional context?

##I have a teensy ++ 2.0 connected working as an automated controller

HookedBehemoth, 2020-06-06:
Please upload the fatal report somewhere else.
It's destroyed only consisting out of ASCII characters.

Santiav90, 2020-06-06:
how can I upload the file?

HookedBehemoth, 2020-06-06:
Zip and attach to a comment here.

Santiav90, 2020-06-06:
[report_000000118b4aca06.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4739962/report_000000118b4aca06.zip)

SciresM, 2020-06-06:
1) This was not made using the latest release. This was made using Atmosphere 0.11.1.

2) The relevant report was generated due to heap exhaustion in ams.mitm during romfs build. 

The error is absolutely not random, if it's real.

I recommend:
a) Actually update to the latest atmosphere release, which you are not using.
b) Provide more details on what behavior triggers your crash report.

Please re-open the issue with new reports if it persists after updating atmosphere.

closed 2020-06-06 by SciresM