Atmosphere-NX/Atmosphere #1031: A fatal error ocurred when running atmosphere

Santiav90, 2020-06-07:
##Hello, me again. The error still happens with atmosphere 12.0. If this can't be fixed, is there a way to make it so that when a fatal error ocurrs, the switch automatically reboots? tbh that would be good enough to fix the problem. The report is the same from yesterday and will be in the comments, or should I attach a new report when the error courrs again?

What's the issue you encountered?
##Every so often after the last update my switch crashes saying "A fatal error ocurred when running atmosphere". I haven't been able to find out when this error happens since it seems random. It also says "Title ID: 010041544d530000" and "Error desc: std::abort() called (0xffe) and report saved to /atmosphere/fatal_errors/report_000000118b4aca06.bin

How can the issue be reproduced?
##havent found a way yet

Crash Report
##

System Firmware Version
##lastest update from yesterday june 5th

Environment?
##I use a fuseeloader and the latest official release

Additional context?
##I have a teensy ++ 2.0 connected working as an automated controller

Santiav90, 2020-06-07:
[report_000000118b4aca06.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4741660/report_000000118b4aca06.zip)

SciresM, 2020-06-07:
You uploaded the same binary you uploaded previously. This is still not an atmosphere 0.12.0 report.

Also, again -- this isn't a random report. The report you uploaded was triggered in romfs building code. What mods are you using/what sysmodules do you have installed? You still did not answer that question that I asked in the last issue, and diagnosis cannot happen without it.

Santiav90, 2020-06-07:
what do you mean mods and sysmodules?
Im sorry im very noob at this. I did mention at the beginning at this report that I was uploading the same crash report as yesterday and asked if I should upload a new one

SciresM, 2020-06-07:
Yes, upload a new one. 

Your report indicates that you are not using a clean atmosphere release. You have add-ins -- game mods or custom system modules that add extra functionality.

What is inside your /atmosphere/contents folder?

Santiav90, 2020-06-07:
ok that is indeed true, I have an extra file in contents made for raid editting in pokemon sword and shield that I cannot get rid of because I want to use it

SciresM, 2020-06-07:
I doubt it is a single file. Your error indicates that you have exhausted the RAM that the romfs building code has available to it.

It has a lot of RAM available to it, and it takes a lot of steps to be memory efficient.

Please fully answer my question about what is inside your /atmosphere/contents folder. I cannot help you if you do not provide me complete diagnostic information.

Santiav90, 2020-06-07:
ok, it has around 9 files, but i have only added two of them, being the pokemon sword one and then an animal crossing one that I can delete since I have no use for it. The other files I have not touched them since I'm afraid that could harm my games. In the next comment I will upload a new report with the contents folder in zip format

SciresM, 2020-06-07:
You are not providing me with the information I need. Take screenshots of your contents folder and upload those.

Santiav90, 2020-06-07:
yes, I said i was going to upload it in the next comment, (the one after this one) since I'm going to wait some minutes until the error happens again so i can provide the report. Here is the screen shot. The one I dont want to remove is 0100ABF008968000
[image]

SciresM, 2020-06-07:
The folders 420000000007E51A and 4100000000000324 are custom system modules.

The first one is tesla. I have no idea what the second one is.

Does the error occur if you remove the second, unknown custom system module? Either way, I grow pretty confident this isn't an atmosphere problem. 

Anyway, could you also post pictures of the /romfs folder inside both the pokemon sword and the animal crossing directories?

Santiav90, 2020-06-07:
yes, how can I reach the /romfs folder?

SciresM, 2020-06-07:
... 0100ABF008968000/romfs for sword. 01006F8002326000/romfs for animal crossing.

Santiav90, 2020-06-07:
[contents.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4741762/contents.zip)
[image]
[image]
[image]

Santiav90, 2020-06-07:
I also included the contents.zip in case its useful

Santiav90, 2020-06-07:
also thanks a lot for helping me with this

SciresM, 2020-06-07:
The 420000000007E51A folder is tesla. The 4100000000000324 folder is apparently "sys-http".

If you trigger the issue on 0.12.0, upload the report and say what you were doing when it occurred.

I would also try removing your custom system modules (by deleting those folders), and seeing if the issue stops happening.

Can't really debug an issue without an 0.12.0 report and/or a consistent way of producing the error.

Santiav90, 2020-06-07:
yep, just finally got the report. Here it is.
[report_00000010eef53318.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4741818/report_00000010eef53318.zip)
Which ones should I delete then?

SciresM, 2020-06-07:
"say what you were doing when it occurred"

Please say what you were doing when it occurred.

`The 420000000007E51A folder is tesla. The 4100000000000324 folder is apparently "sys-http".`

Remove both of them, I guess.

Santiav90, 2020-06-07:
I'm not sure what was going on when it happened, since I have a teensy ++ 2.0 repeating a loop of actions over and over. So its not something that is guaranteed to happen, it just happens every hour or so. If I am able to catch the moment it happens I will mention it here. I've left it going for around 40 minutes and there hasn't been an error yet.

Santiav90, 2020-06-07:
I removed both of them and the error still ocurred. I think it happens when closing and opening the game, I will watch it for an hour to make sure when it ocurrs

Santiav90, 2020-06-07:
Is there a way to make it so that if there is an error it just reboots?

Santiav90, 2020-06-07:
@SciresM ?

SciresM, 2020-06-07:
I sleep between ~3:30 am my time and ~6:30 am my time. Tagging me is not helpful.

Error is again out of memory during romfs build. Maybe there's a memory leak of some kind in the romfs stuff.

Auto reboot after fatal error --> not in fusee.

What is your automation program doing? Infinitely closing/reopening a game?

Santiav90, 2020-06-07:
yeah, basically doing some stuff in the game and then closing it and opening it

Santiav90, 2020-06-07:
sorry

Santiav90, 2020-06-07:
This time I was able to see when the error ocurred and it was just while starting a raid on pokemon.

benbooth493, 2020-07-09:
I am experiencing the same issue exactly.
I have a clean atmosphere and sept folder. Nothing in there from my old Kosmos install.

SciresM, 2020-07-09:
@benbooth493 This issue is that there's some memory leak in romfs building code so if you automate closing/opening then after a while memory fragmentation causes an oom --> crash.

I...sincerely doubt your issue is the same.

benbooth493, 2020-07-09:
So I have exactly the same error message. I accidently allowed the system to upgrade to 10.0.3. So I ripped out Kosmos and replaced with atmosphere.

SciresM, 2020-07-09:
"exactly the same error message" <-- does not mean this is at all related to this issue. All atmosphere errors give that error message.

Make a new issue and upload your report. I'm entirely sure your issue is unrelated to this.

kasboy82, 2020-07-28:
Hi, I'm having the same problem
[report_000000000635a151.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4989551/report_000000000635a151.zip)

HookedBehemoth, 2020-07-28:
@kasboy82 
exFAT card without FAT32 only firmware.
Format your sd card to FAT32.

kasboy82, 2020-07-28:
> 
> 
> @kasboy82
> exFAT card without FAT32 only firmware.
> Format your sd card to FAT32.

thanks. can it be fixed if I change on choujonoirNX to firmware with exfat?

DerinChiu, 2020-12-21:
Solved this problem by #https://github.com/Atmosphere-NX/Atmosphere/issues/813#issuecomment-696612167

giannimk3, 2021-01-04:
Hello, I have a similar error. I was on 10.0.2. and atmosphere 0.14.4 on my exfat EmuMMC, then I updated atmosphere to 0.16.2 and used daybreak to update to FW 11.0.1. I chose the (fat32+exFAT) option. 

Since then, my Atmosphere crashes with: 
A fatal error occured when running atmosphere. 
Program ID: 0100000000000bd00
Error desc: std::abort() called (0xffe)
You can find the .bin report in the file. I hope someone can help me.
[report_000000000aa731fd.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5766724/report_000000000aa731fd.zip)

SciresM, 2021-01-04:
Your issue is completely unrelated. Update mission control, and this isn't a support forum.

WakamoleEpico, 2021-03-23:
hola que tal, me dio un problema parecido es este "a fatal error occured when running atmosphere
program ID: 01004154d530000
error desc: std: :abort() called (0xffe)
report saved to/atmosphere/fatal_errors/report_000000000b44e679.bin"
que puedo hacer?
gracias de antemano

Ammako, 2021-07-08:
Basically the same as OP, but different mods. Closing the game and reopening to reset several times, from time to time it crashes like this on reopening.

Using No Outlines mod, and ReTextured 1.6

https://yuzu-emu.org/wiki/switch-mods/

(Yuzu page, the mods work fine on console though. Although the retexture mod is 1 GB so that could be cause for error.)

Otherwise I have Fizeau sysmodule, and home menu theme. Nothing else.

[fatal_errors.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6786117/fatal_errors.zip)

SciresM, 2021-07-08:
How many times is several?

Ammako, 2021-07-08:
Half a dozen to a dozen or so

SciresM, 2021-07-08:
What game? Pokémon sword?

Ammako, 2021-07-08:
Shield

BeniyashaSSB, 2021-07-09:
I'm having the same problem as OP and a couple other people as well. I have a few texture mods and one "custom mod" that only alters specific Pokemon's encounter data, and my switch encounters a panic error nearly identical to OP's after closing and opening Pokemon Sword about a dozen times. This is after I have installed Atmosphere 0.19.5 and Hekate 5.5.8 earlier today.

Specific mods include: Skin mods for 5 Pokemon, 3 Clothing specific mods, and 1 mod created via pkNX that allows for specific Pokemon encounters to have their "shiny lock" removed. (can link the mods in particular if necessary)

I have a lot of folders in my atmosphere/contents folder that have cheat files for games I don't own, but for the most part, I haven't done a lot of modding except for the Tesla menu and a few other specific games. I don't have any themes installed, nor have I intently put any additional sysmodules on my switch due to my not wanting to mess with it.

nota-debug, 2021-07-11:
Can anyone help me my baby cousin updated my switch a while ago and after that it stop working and gave the same error. my sd card is in fat32 I have tried to set the bit archive but it didnt work either. I will post the error bin file below. Thank you for your cooperation

nota-debug, 2021-07-11:
[report.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6796895/report.zip)

masagrator, 2021-07-11:
> [report.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6796895/report.zip)

Delete missioncontrol or update it.

nota-debug, 2021-07-11:
im a noob can you give me steps please and thank you

nota-debug, 2021-07-11:
Ok so i moved it to downloads and it still gave me the same error mesage i will post the bin file below

nota-debug, 2021-07-11:
[New WinRAR ZIP archive.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6796919/New.WinRAR.ZIP.archive.zip)

nota-debug, 2021-07-11:
@masagrator

nota-debug, 2021-07-11:
@SciresM

masagrator, 2021-07-11:
Delete `contents`  folder and try again.
and don't ping because you can't wait even 30 minutes for reply...

nota-debug, 2021-07-11:
tysm it worked

nota-debug, 2021-07-16:
[report_00000000088c75bc.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6832765/report_00000000088c75bc.bin.zip)
same problem happened :\

nota-debug, 2021-07-16:
@masagrator 
sorry lol

nota-debug, 2021-07-16:
@SciresM

SciresM, 2021-07-16:
@nota-debug do not ping me for stuff like this, this is not a support forum.

You have archive bits set on atmosphere subdirectories. Run hekate's archive bit fixer, and if that doesn't work delete atmosphere folder/extract cleanly from latest release zip using not-macOS.

SciresM, 2021-07-16:
Re-opening because I forget this is the memory leak in romfs building issue that this other guy is hijacking incorrectly.

fredfrog6, 2021-09-08:
I'm not sure if this is the exact same issue as OP but I'm having an issue that seems to be exhibiting the same symptoms. I receive the same error described by OP every so often when I open a title. The only mods I'm running are outline removal patches for pokemon swsh and I've removed them in the past and still experienced the same crashes. Attached is the crash report. Any help would be appreciated.

[report_00000001cceddd95.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7125436/report_00000001cceddd95.zip)

SciresM, 2021-09-08:
@fredfrog6 yes, this issue is the same -- it's some kind of memory leak triggered by repeatedly launching game-with-big-romfs-and-mods. It's been on my to-do list for a while.

[2 more replies]

commit ab81ed27 (2021-09-21): fs.mitm: fix memory leak in romfs build (closes #1031)
closed 2021-09-21 by SciresM