Atmosphere-NX/Atmosphere #1035: Wake from sleep crash

icwienerr, 2020-06-12:
**Bug Report**

### What's the issue you encountered?

Atmosphere crash when wake from sleep 

### How can the issue be reproduced?

When i leave the switch in sleep mode for a long time crash when wake from sleep mode.

### Crash Report

Result:                          0x222A02 (2002-4373)

Process Info:
    Process Name:                sdb
    Program ID:                  0100000000000039
    Process ID:                  000000000000007a
    Process Flags:               00000137
    User Exception Address:      0000003a3d446000
Exception Info:
    Type:                        User Break
    Address:                     0000002d4547acc4 (sdb + 0x7acc4)
    Break Reason:                0x0
    Break Address:               0000004c2f321810
    Break Size:                  0x4

[crash_reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4770173/crash_reports.zip)
[crash_report dumps.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4770174/crash_report.dumps.zip)
[fatal_reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4770175/fatal_reports.zip)
[fatal_reports dumps.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4770176/fatal_reports.dumps.zip)

### System Firmware Version

10.0.4</br>

### Environment?

Bootloader: hekate 5.2.1
AMS 0.120

spling-du-futur, 2020-06-12:
try to use fusee-primary

SciresM, 2020-06-28:
4373 is a save data verification issue. This seems like random-error at first blush?

closed 2020-06-28 by SciresM