Atmosphere-NX/Atmosphere #1040: Atmosphere crashing On launch of Minecraft Dungeons

jayvenem, 2020-06-16:
## Bug Report
* 
### What's the issue you encountered?

[ Whenever i Open Minecraft dungeons, it starts for 5 seconds and then atmosphere crashes. It gives me error 2168-0002. Before that I was just playing splatoon 2 with a friend. ]</br>
[ Yes. ]</br>
[ I updated atmosphere and hekate to see if it will fix it but no use. I deleted the folder for themes and checked to see if my edizon cheats were automatically on, in which they weren’t. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Launch Minecraft dungeons on cfw sysnand, that’s all. ]</br>

### Crash Report

[https://gist.github.com/jayvenem/ea31c4058e6f8aad33c92532d9a34efb ]</br>

### System Firmware Version

10.0.4</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  - [ Hekate 5.3.0  ]
- Do you have additional kips or sysmodules you're loading: Sys-Tune, SaltyNX, emuiibo and Sys-Con

### Additional context?

- Additional info about your environment:
- [ N/A ]

HookedBehemoth, 2020-06-16:
The crash report you sent is from the edizon overlay.
Unless this is the wrong report.

jayvenem, 2020-06-16:
> The crash report you sent is from the edizon overlay.
> Unless this is the wrong report.

It is? I got it from the crash reports in the atmosphere folder

jayvenem, 2020-06-16:
> The crash report you sent is from the edizon overlay.
> Unless this is the wrong report.

Ah, apologies I found the real report. I fixed the post

HookedBehemoth, 2020-06-16:
Indirectly caused by sys-tune.

Remove the boot2.flag from /atmosphere/contents/4200000000000000/flags

The new version let's you toggle it on the fly without rebooting.

jayvenem, 2020-06-16:
> Indirectly caused by sys-tune.
> 
> 
> 
> Remove the boot2.flag from /atmosphere/contents/4200000000000000/flags
> 
> 
> 
> The new version let's you toggle it on the fly without rebooting.

Thank you! It works now!

closed 2020-06-16 by jayvenem