Atmosphere-NX/Atmosphere #1041: Could not boot when chainloaded with Hekate 5.3.0

dothackjhe, 2020-06-16:
## Bug Report

### What's the issue you encountered? 

The console goes into "black screen" after the Atmosphere logo when chainloaded with the latest iteration of Hekate (5.3.0).</br>
The issue arose when I upgraded to Atmosphere 0.13.0.</br>
The system boots fine when I reverted back to Atmosphere 0.12.0.</br>

### How can the issue be reproduced?

Chainload Atmosphere 0.13.0 with Hekate 5.3.0.</br>

### System Firmware Version

10.0.2</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate 5.3.0
- Official release or unofficial build: Official build

HookedBehemoth, 2020-06-16:
Assuming you mean launch fusee-primary by chainload:
> fusee-primary was last updated in: 0.13.0.

Did you update it from where you chainload?

dothackjhe, 2020-06-16:
Yes, I did. @HookedBehemoth

dothackjhe, 2020-06-16:
I do not think I was the only one who encountered it, though: https://gbatemp.net/threads/atmosphere-nx-custom-firmware-in-development-by-sciresm.496832/page-277#post-9096844

It's just funny no one made a report about it.

HookedBehemoth, 2020-06-16:
Piracy is not supported.

dothackjhe, 2020-06-16:
What if I say that I only run homebrews, will you even believe it?

Also, what made you think that I use Hekate for piracy purposes? Does it have something to do with Atmosphere?

MichaelRebirth, 2020-06-16:
Im having the same problem chainloading it error code was 010notsurehowmany03 after passing the sept logo.

ELY3M, 2020-06-16:
just use fuse-primary like me.

xmun2k, 2020-06-18:
Having the same problem with fusee-primary. Using a completely clean install even on a new SDCard with fresh eMMC partition created.

KiraPT, 2020-06-27:
I had the same issue and after some reading and some work I was able to fix the issue by using the fix from Hekate and rebuilding the hekate_ipl.ini file. 
Now it boots SysNand, EmuNand and Stock with no issues. 

If it is any help, here's my hekate_ipl.ini file:

[hekate_ipl.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4822468/hekate_ipl.zip)

************
[config]
autoboot=2
autoboot_list=0
bootwait=2
backlight=100
autohosoff=0
autonogc=1
updater2p=0

[Atmosphere (SYSNAND)]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
emummc_force_disable=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
{}

[Atmosphere (EMUMMC)]
payload=atmosphere/fusee-primary.bin
kip1patch=nosigchk
emummcforce=1
logopath=bootloader/bootlogo_emunand.bmp
icon=bootloader/res/icon_payload.bmp
{}

[Stock]
fss0=atmosphere/fusee-secondary.bin
stock=1
emummc_force_disable=1
{}
*****************

SciresM, 2020-06-28:
This particular issue was solved by ctcaer using a fix I believe, the other one with xmun is maybe different.

closed 2020-06-28 by SciresM

dothackjhe, 2020-06-28:
I actually had it resolved by deleting all the folders which pertains to the older version of Atmosphere and replacing everything with the ones that come with Atmosphere 0.13.0 as well as the latest sigpatch. As it appears, there are some leftover files from 0.12.0 that was in conflict with 0.13.0 that prevents the system from booting properly.