Atmosphere-NX/Atmosphere #1044: ATMOSPHERE 0.13.0 IS CRASHING ON BOOT,

spling-du-futur, 2020-06-17:
## Bug Report

### What's the issue you encountered?

Simple, AMS crash when console boot</br>
i updated ams 0.12.0 to 0.13.0</br>
i have a nxtheme installed on 0.12.0</br>

### How can the issue be reproduced?

boot the console</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

9.2.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  
- Do you have additional kips or sysmodules you're loading ?
Tesla<br/>
ldn_mitm<br/>
sys-con<br/>
- Homebrew software installed: 
awoo installer<br/>
goldleaf<br/>
nxtheme installer<br/>
kosmos toolbox<br/>
hb app store<br/>

spling-du-futur, 2020-06-17:
i switch to ams 0.12.0

spling-du-futur, 2020-06-17:
only for remove nxtheme

SciresM, 2020-06-17:
can't reproduce, include crash reports/fatal reports :P

Also test without your sysmodules.

closed 2020-06-17 by SciresM