Atmosphere-NX/Atmosphere #1046: "A fatal error occurred when running Atmosphere"

Jimbo420, 2020-06-17:
## Bug Report

### What's the issue you encountered?

My switch gives this error everytime it try to load into atmosphere either from the emummc or fusee primary. I had not used the switch in a few days and it was in sleep mode. When i used it, it just had the error on screen so it might have been like that for a few days without me noticing.

### How can the issue be reproduced?

I was playing a game last time i used the switch. 

### Crash Report

https://gist.github.com/Jimbo420/94554a86a2a68b7dd6109e9ea54304a1

### System Firmware Version

10.0.4 now but it was 10.0.3 when the error first happened. Tried to fix it by updating.

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee primary using Hekate
- Official release or unofficial build: Official release version 5.3.0
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Tinfoil

### Additional context?

- Additional info about your environment: I was using slightly older version of Hekate and Fusee primary when it first happened but i updated it to see if that fixed it but it didnt change anything. Also my emummc is on version 9.1.0.
[image]
[image]

SciresM, 2020-06-17:
...upload the fatal error binary...?

Jimbo420, 2020-06-17:
Sorry lol 
https://mega.nz/file/KT5zWJyL#61AWugVSit-MHWcRI4LF-8EryIjPIop04vDqj2sHaVw

SciresM, 2020-06-28:
Your system settings savedata is corrupted (region code has uninitialized bit set).

This isn't really an atmosphere problem (this is indeed a fatal error), fix on your end will be to delete your system settings savedata. Reconstruction of system savedata should fix it (unless you corrupted ProdInfo or something).

Try deleting 800...0050 save from SYSTEM.

closed 2020-06-28 by SciresM

erwanaitairforce, 2022-03-04:
Hello, I have exactly the same problem, but I did not really understand your explanation, could you help me, thank you very much.
[image]
[image]

erwanaitairforce, 2022-03-04:
[image]

erwanaitairforce, 2022-03-04:
[image]

SciresM, 2022-03-04:
Don't make a new issue. Just follow the advice given by the text on the console.

erwanaitairforce, 2022-03-04:
How I Write new issue, please. (Sorry i’m a noob😅)

erwanaitairforce, 2022-03-04:
@SciresM  ok, but I do not know how to do: (try Running an archive bit fixer tool).