Atmosphere-NX/Atmosphere #1051: Atmosphere 0.13.0 prevents anything from loading

armoar334, 2020-06-18:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

after updating to 0.13.0, any modded games / title take over will get stuck on the 'nintendo switch' splash. any unmodded games will boot, but refuse to go further than the opening loading screen i.e animal crossing gets stuck on the little island and black screen. attempting to exit to the home menu brings me to a blank grey background, but for a short while the power menu and joycon pair/ attach overlay will work. i also cannot re-enter the games from here.

### How can the issue be reproduced?

installing atmosphere 0.13.0

### Crash Report

N/A

### System Firmware Version

10.0.4

### Environment?

Hekate 5.3.0
Official release
emuiibo
HBappstore, jksv, nxthemes, awoo, goldleaf,
### Additional context?

SD slot occasionally fails, but booted atmosphere multiple times and it was working each time

SciresM, 2020-06-18:
Does issue occur on stock atmosphere without add-ins? I can't reproduce -- mods are working fine here taking over Pokemon Sword.

armoar334, 2020-06-18:
Yes, it does

SciresM, 2020-06-18:
What makes you confident that it's atmosphere and not your mod?

Again, I cannot reproduce -- mods work totally fine here.

armoar334, 2020-06-18:
because every mod i have on worked on 0.12.0 and its not just mods: any cfw related thing, including title takeover refuses to boot

SciresM, 2020-06-18:
That sounds not like atmosphere, because atmosphere clearly works in most cases. That sounds like your setup.

Delete your atmosphere folder. Clean install stock atmosphere with no add-ins at all using latest release zips. Report results.

armoar334, 2020-06-18:
fresh install seems to have fixed it. thx!

closed 2020-06-18 by armoar334

cheesycod, 2020-07-12:
> goldleaf

Piracy is not supported here

armoar334, 2020-07-12:
A, none of my games are pirated, that is for forwarders
B. This issue was closed nearly a month ago, why open it to post something that is completely irrelavent

fincs, 2020-07-12:
Piracy is not supported. Installing "forwarders" entails using the same infrastructure (sigcheck patches) needed for piracy (which then entails Nintendo bans on consoles).

Users are intented to exclusively use nx-hbloader and nx-hbmenu to run homebrew.