Atmosphere-NX/Atmosphere #1053: Atmosphère 0.13.0 blank screen after Sept splash screen.

xmun2k, 2020-06-19:
### What's the issue you encountered?

After Sept splash screen, Switch display goes blank and remains that way. Pressing the power button causes the process to loop.

### How can the issue be reproduced?

1) Download Atmosphere
2) Copy contents to SD card
3) Boot to RCM
4) Send fusee-primary payload

### Crash Report

No crash reports were created.

### System Firmware Version

8.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: **official fusee-primary 0.13.0**
- Official Atmosphère 0.13.0

### Additional context?

- Tried two SD cards

1. 256GB Samsung Evo
2. 64GB Intenso

- Tried completely clean installation with no modules or patches completely vanilla.
- Tried starting from fresh eMMC

- Restoring to Atmosphère 0.12.0 solves the issue.

xmun2k, 2020-06-19:
Adding a video link to show you what's going on. Off-screen, the switch is in RCM mode connected to my PC and I push fusee-primary (latest) video TegraRCM.

After the video ends, the screen remains blank.

https://youtu.be/TvA96APXa9g

hrvylein, 2020-06-19:
Have you updated all sysmodules? Release notes say every sysmodule has to be compiled against 0.13.

masagrator, 2020-06-19:
> Have you updated all sysmodules? Release notes say every sysmodule has to be compiled against 0.13.

Not every. Every using atmosphere-libs as core part.

KiraPT, 2020-06-20:
Having the exact same issue with my switch, but if I go back to 0.12.0, it solves the issue.

xmun2k, 2020-06-20:
> Have you updated all sysmodules? Release notes say every sysmodule has to be compiled against 0.13.

I am using a completely vanilla installation. No sysmodules beyond what would be included with the release.

xmun2k, 2020-06-20:
> Having the exact same issue with my switch, but if I go back to 0.12.0, it solves the issue.

Did you watch the youtube video I linked? Is it exactly the same ?

KiraPT, 2020-06-20:
> 
> 
> > Having the exact same issue with my switch, but if I go back to 0.12.0, it solves the issue.
> 
> Did you watch the youtube video I linked? Is it exactly the same ?

Yes. It happens exactly the same.

haganebr, 2020-06-21:
Same thing happened to me. Going back to 0.12 fixed it.

SwiftMove, 2020-06-22:
The issue is with hekate, not Atmosphere. CTCaer has posted a fix in this thread...

https://github.com/CTCaer/hekate/issues/460#issuecomment-647019768

xmun2k, 2020-06-22:
> Same thing happened to me. Going back to 0.12 fixed it.

Could I ask you what FW version your SYSNAND is on?

xmun2k, 2020-06-22:
> > > Having the exact same issue with my switch, but if I go back to 0.12.0, it solves the issue.
> > 
> > 
> > Did you watch the youtube video I linked? Is it exactly the same ?
> 
> Yes. It happens exactly the same.

Could I ask you what FW version your SYSNAND is on?

haganebr, 2020-06-22:
> > Same thing happened to me. Going back to 0.12 fixed it.
> 
> Could I ask you what FW version your SYSNAND is on?

SysNAND 8.1.0
EmuNAND 10.0.4
I don't even have hekate on my system, so that couldn't be the issue.

xmun2k, 2020-06-22:
> > > Same thing happened to me. Going back to 0.12 fixed it.
> > 
> > 
> > Could I ask you what FW version your SYSNAND is on?
> 
> SysNAND 8.1.0
> EmuNAND 10.0.4

Same as me so. Might be an issue with SysNAND 8.1.0

KiraPT, 2020-06-22:
> 
> 
> > > > Same thing happened to me. Going back to 0.12 fixed it.
> > > 
> > > 
> > > Could I ask you what FW version your SYSNAND is on?
> > 
> > 
> > SysNAND 8.1.0
> > EmuNAND 10.0.4
> 
> Same as me so. Might be an issue with SysNAND 8.1.0

My SysNAND is 10.0.4 and my emuNAND is 10.0.4 and the issue is the same. But if I go back to 0.12.0 the issue is fixed. As I've read, it appears the issue is with Hekate. I will test the fix that was released and see if the issue is solved

SciresM, 2020-06-28:
Turns out fusee/exo2 don't work on units which use new device key generation. Fixed locally, testing with xmun before merge.

commit 1473adf5 (2020-06-28): fusee/exo: correct device key management for newer consoles (closes #1053)
closed 2020-06-28 by SciresM