Atmosphere-NX/Atmosphere #1055: Crash on launching homebrew menu once in CFW

SuperIronBob, 2020-06-20:
## Bug Report

### What's the issue you encountered?

Trying to do initial set up of CFW using switch.homebrew.guide . I can get into the stage for booting from RCM to Atmosphere, but once there the homebrew menu crashes when opening.

I've tried with completely clean new downloads of:
-  recommended defaults+pegascape from sdsetup
- hekate+atmosphere from releases on respective github 
- just atmostphere from releases on github with fusee delivered over RCM

Each time I do so I get an error 2011-0301 (Remote Process is Dead) for program: 0100000000000023 no report is generated in any of the reports directories, however the fatal_errors directory was created. 

### How can the issue be reproduced?

1. clean SD card set up
2. copy files from atmosphere releases (I used the latest official release against 3295132)
3. Boot from RCM using associated fusee
4. attempt to open album
5. reach mentioned fatal error

### Crash Report

No report is generated in any of the reports directories, however the fatal_errors directory was created. If there's something else I can do to get the report I'd be glad to.</br>

### System Firmware Version

4.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  - hekate and fusee were both tried
- Official release or unofficial build: 
  - Official release version 0.13.0
- No other modifications were done

### Additional context?

console is pretty stock, it was new in box up till about a week ago.

HookedBehemoth, 2020-06-21:
Seems like my npdmtool changes cause fs to crash on 4.1.0.

hbl built with 1.6.1 switch-tools works fine.

I've not seen reports that this affects any other system version.

vahnLeiden, 2020-07-05:
I am going through the same issue. Did you find any workaround for this? Thanks!

HookedBehemoth, 2020-07-05:
Downgrade hbl

SciresM, 2020-07-08:
@HookedBehemoth this actually affects *every* system version <= 4.1.0.

One thing I'm concluding from this issue is that nobody runs <= 4.1.0.

SciresM, 2020-07-09:
Fixed by https://github.com/switchbrew/switch-tools/pull/29 and https://github.com/switchbrew/nx-hbloader/pull/30.

Atmosphere 0.14.0 will include a fixed hbl binary.

closed 2020-07-09 by SciresM

impeeza, 2020-07-10:
> 
> 
> @HookedBehemoth this actually affects _every_ system version <= 4.1.0.
> 
> One thing I'm concluding from this issue is that nobody runs <= 4.1.0.

For me is a problem because I have a Switch untouched on 4.1.0 and emuMMC, recently i did broke my SD card, and try to recreate all from scratch, and unable to run new atmosphère version on the untouched sysnand.  so I did have to firts use a old atmosphère for creating a emunand dump, then update the emunand to latest firmware then update all to the new atmosphère version.