Atmosphere-NX/Atmosphere #1056: Error 20001-0123 when pokémon sword in-game

amine4567, 2020-06-20:
## Bug Report

### What's the issue you encountered?

I updated atmosphere to 0.13.0 and the switch firmware to 10.0.4 and now Pokémon Sword crashes seconds after going in-game.

### How can the issue be reproduced?

- Boot into Atmosphere 0.13.0 using fusee-primary
- Launch Pokémon Sword 
- Go in-game
- Crashes 1 second in (Error 20001-0123).

### Crash Report
2 crash reports at the same time:
https://gist.github.com/amine4567/60d3cfaac8c1c18f15b02fe0b7695fa5

### System Firmware Version

10.0.4

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee-primary
- Official release or unofficial build: Official release
- Homebrew software installed: [ * ]

### Additional context?

N/A

closed 2020-06-20 by amine4567

cheesycod, 2020-07-12:
Spam?

masagrator, 2020-07-12:
> Spam?

Your post is spam if you're replying to issue closed 22 days ago by OP.