Atmosphere-NX/Atmosphere #1057: Crashes while playing Pokémon: "A fatal error has occured".

PP-theSLAYER, 2020-06-21:
## Bug Report

### What's the issue you encountered?

The game suddenly freezes, and leads to a crash.
Seems to occur when I load a different area, enter a raid, check the dex etc.

### How can the issue be reproduced?

To be honest, seems to occur randomly (besides loading a new area or simply performing an action). Only started happening after the update 1.2.0. Did not occur for previous updates.

### Crash Report

[01592633019_0100abf008968000.log](https://github.com/Atmosphere-NX/Atmosphere/files/4808933/01592633019_0100abf008968000.log)
This is the most recent more. I have more and can attach them if need be.

### System Firmware Version

10.0.4

### Environment?

- fusèe
- Official release 
  - 0.13.0
- No additional kips or sysmodules (I delete the Atmosphere folder before making performing a fresh install)
- Homebrew software installed: [All via HBL: Checkpoint, ChoiDuJourNX, Firmware Dumper, EdiZon, nxdumptool, Reboot to Payload, WAINCartDumper]

### Additional context?

- No EmuMMC, SD 32GB, Class 10, ExFAT, no autosave on Pokémon.

raytan1807, 2020-06-24:
What's the issue you encountered?
The game suddenly freezes, and leads to a crash.

How can the issue be reproduced?
During the conversation with Mustard after obtaining the Max Honey (both sword & shield ver). Refer picture.
[image]

Crash Report
[01593018546_01008db008c2c000.log](https://github.com/Atmosphere-NX/Atmosphere/files/4827089/01593018546_01008db008c2c000.log)

System Firmware Version
9.2.0 & 10.0.4

Environment?
Hekate 5.3.0
Official release
0.13.0
No additional kips or sysmodules
Homebrew software installed: [All via HBL: Checkpoint, ChoiDuJourNX, JKSV, EdiZon, Reboot to Payload]

Additional context?
EmuMMC, 256GB, Class 10, Fat32, no autosave on Pokémon, no Auto RCM.

SciresM, 2020-06-28:
Error code is "sdmmc driver (can't perform voltage switch)", so I'm going to close this.

closed 2020-06-28 by SciresM