Atmosphere-NX/Atmosphere #1058: Use cfw when sd card reader is broken/malfunctioning/dying/not working as intended and cant be replaced and when switch is banned.

Expertsonicgamer32x, 2020-06-22:
## Use cfw without sd card

### I want to use cfw without an sd card
#### Overview:
- [ since my switch is banned and the sd card reader is doing a terrible job at reading it without failing when moving my switch I want to use cfw without an sd card but i found no luck of doing it. I'm young and I don't have the resources to but my own stuff online yet and unless there are ways to buy stuff online on online retails with a gift card instead of a credit card to ship to a house I want to use sigpatches and mods with my switch internal storage please make a build where it uses te switch internal storage instead of an sd card it would be really helpful.  ]</br>

#### Smaller Details: 
- [ Make the switch use the internal storage instead of the sd card storage ]</br>

SciresM, 2020-06-22:
Atmosphere will almost certainly never support no-sd-card boot.

closed 2020-06-22 by SciresM

Expertsonicgamer32x, 2020-06-22:
i was wondering.