Atmosphere-NX/Atmosphere #1060: Ninjala broke the Fan

Lost0Neko, 2020-06-26:
## Bug Report

Hey,

Yesterday i Started Ninjala, after that the switch froze and now the fan doesn't work in sysCFW and Stock mode.
Normally i would assume the fan broke, BUT the fan does work, in the hekate menu...

Detailed:
1. I turned on my Switch, played smash for like 2 hours (Switch fan works fine and is "cold")
2. I closed Smash and Opened Ninjala (First ever time)
3. The Switch froze
3.1 It was in the screen where the Game openes the Browser (So i could accept the Terms of service)
4. I soft reset my switch be cause i couldn't do ANYTHING else
5. Same thing happened
6. Started my switch again, but this time in Stock mode
7. It worked this time, but i noticed that the fan didn't work
8. I rebooted my switch in Cfw and checked with the Status monitor if the fan was "working", it said its running at 100% but the fan was silent and no air came out
9. The switch overheated

Switch Firmware: (Newest) 10.0.4
Atmosphere Firmware: 0.12.0

Any detail's i missed?

SciresM, 2020-06-26:
Re-open using the issue templates.

closed 2020-06-26 by SciresM

masagrator, 2020-06-26:
And use the newest Atmosphere before opening new issue. :P

cheesycod, 2020-07-12:
It sounds like a hardware problem

cheesycod, 2020-07-12:
> Started my switch again, but this time in Stock mode
If it happens in OFW as well, it's not our fault

ehx-v1, 2020-08-16:
> > Started my switch again, but this time in Stock mode
> 
> If it happens in OFW as well, it's not our fault

It does sound like CFW fried something permanently, though.