Atmosphere-NX/Atmosphere #1062: AMS 0.13 - Local multiplayer : Software was closed because an error occured

Sblawx, 2020-06-28:
## Bug Report

### What's the issue you encountered?

First discovered this error when trying to play ACNH in local multiplayer on AMS 0.13.0. When the console starts opening the local wireless service, it shows the following error message : The software has stopped because an error occured." When you click OK, the game quits. Some of the games like pokemon sword/shield are automatically triyng to connect after loading a save. So the error happens and the games are unplayable. 

I fixed the issue by rolling back to Atmosphere 0.12.0.</br>

### How can the issue be reproduced?

- Clean SD card set up
- Install hekate 5.3.0 and atmosphere 0.13.0 with fusee-primary.bin
- Launch your emunand with fusee-primary
- Install games that have local multiplayer function (ACNH or Pokemon)
- Try to access local multiplayer in ACNH / Just launch pokemon and you will have the error when wireless logo appear</br>

### Crash Report

https://gist.github.com/Sblawx/ec4c07d96445d3d692911b26c32249cc</br>

### System Firmware Version

10.0.4|AMS 0.13.0|E</br>

### Environment?

- Hekate 5.3.0 - Fusee-primary
- Atmosphere 0.13.0 official release
- ldn_mitm 1.3.4 (siwtched off)
- sys-ftpd-light

### Additional context?

[N/A]

SciresM, 2020-06-28:
`ldn_mitm 1.3.4 (siwtched off)
` "Switched off" what?

You can't switch off a mitm module.

Does the error occur without that?

Sblawx, 2020-06-28:
When I said switched off, I mean this : 

[image]

But it does the same if switched ON

SciresM, 2020-06-28:
Remove ldn_mitm entirely. Does the issue still occur?

SciresM, 2020-06-28:
(Also test without sys-ftpd-light, if that fails)

Sblawx, 2020-06-28:
Well thanks a lot for your help, it seems that ldn_mitm needs to be updated to work with AMS 0.13.0. 
I'll stay on AMS 0.12.0 until an update comes for ldn_mitm, I'm using it to play online with friends.

closed 2020-06-28 by SciresM