Atmosphere-NX/Atmosphere #1064: Crashing with new horizons and smash

jayvenem, 2020-06-29:
## Bug Report
* 
### What's the issue you encountered?

[ Whenever i Open Smash or Horizons it causes my atmosphere to crash. It gives me error 2168-0002. Before that I was just playing sonic mania. ]</br>
[ Yes. ]</br>
[ I updated atmosphere and hekate to see if it will fix it but no use. I deleted the folder for themes and checked to see if my edizon cheats were automatically on, in which they weren’t. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Launch Either Smash or New horizons on cfw sysnand, that’s all. ]</br>

### Crash Report

[https://gist.github.com/jayvenem/c045ccce632bf42691d838076f425c27 ]</br>

### System Firmware Version

10.0.4</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  - [ Hekate 5.3.0  ]
- Do you have additional kips or sysmodules you're loading: SaltyNX, emuiibo and Sys-Con

### Additional context?

- Additional info about your environment:
- [ N/A ]

SciresM, 2020-06-29:
The crash is in saltysd. Remove it.

closed 2020-06-29 by SciresM

jayvenem, 2020-06-29:
> The crash is in saltysd. Remove it.

No luck

SciresM, 2020-06-29:
> > The crash is in saltysd. Remove it.
> 
> No luck

Upload new crash report then.

Your report states the crash is in saltysd:
```
        PC:                      0000000138468db8 (saltysd_core + 0x2db8)
```

masagrator, 2020-06-29:
Smash was reported crashing with SaltyNX with newest SaltyNX versions (which looks like people doesn't care that much because noone want to make normal issue). But it shouldn't be the case with Animal Crossing.

SciresM, 2020-06-29:
@masagrator Given the whole "multiple sysmodules" thing, I'm gonna guess it's one of them (especially as I play a few hours of animal crossing on my ams console in any given day).