Atmosphere-NX/Atmosphere #1066: Fatal error titleid: 0100000000000005

iixela, 2020-07-01:
Hi!

I Update from 9.0.1 to 10.0.4 and i can load payloads (heckate, primary or reinx) but show black screen after.
 
I used a format sd card in exfat and with all new folders. 

Thanks a lot for you support!

Fatal error reports:

http://www.mediafire.com/file/timvnwnpt6rq1ly/report_0000000001c91b68.bin/file

http://www.mediafire.com/file/xg07i0yjmlaeapa/report_000000000128b2f7.bin/file

http://www.mediafire.com/file/uh8dmzubnr8n2e7/report_0000000000c3933a.bin/file

SciresM, 2020-07-01:
You didn't update properly.

Your console does not have 10.x package2 in emmc; 10.x kernel had a breaking API change, and so the boot sysmodule is erroring trying to use the new API.

Manually write 10.x package2 to your emmc (or go to a support discord for help fixing stuff).

closed 2020-07-01 by SciresM

Chrscool8, 2020-07-05:
Just thought I'd drop a mini walk-through of how to do that fix since I had a lot of trouble until I finally figured it out. My emuMMC was updated to 10.0.4 via ChoiDujourNX and I got only fatal errors on start until the following. _Your mileage may vary wildly and I'm sure there are many different ways to go about the same procedure. Do at your own risk and protect that backup!_

To expand on what SciresM said above: I needed to update my (clean, internet capable) SysNAND to 10.0.4, then grab a file from its NAND, **'BCPKG2-1-Normal-Main.bin'**, and inject that over one in my existing emuMMC NAND.

- Getting Started
1) Update Hekate and Atmosphere to the latest releases on the SD Card.

2) RCM mode into Hekate

[image]

3) Go to the tools menu

[image]

4) Go into USB Tools

[image]

- Back up the current emuMMC **'BCPKG2-1-Normal-Main.bin'**.

5) Mount your current emuMMC NAND by clicking '_emu RAW GPP_'.

6) Get and run *NxNandManager v4.0* and open your newly mounted drive

[image]

7) Save your current **'BCPKG2-1-Normal-Main.bin'** somewhere safe on your computer

[image]

8) Unplug and replug the cable.

- Get **'BCPKG2-1-Normal-Main.bin'** from OFW eMMC NAND

9) Back in the USB Tools, this time mount '_eMMC RAW GPP_'.

[image]

10) Relaunch and remount the drive in *NxNandManager v4.0*, and once again copy the eMMC's **'BCPKG2-1-Normal-Main.bin'** to your PC

[image]

11) Unplug your cable and replug it to eject

- Injecting the OFW eMMC's file to the CFW emuMMC's nand.

12) Third and final move! This time, turn off 'Read-Only' at the bottom. Move carefully now, things could go bad if you mess up! Once more, mount '_emu RAW GPP_'.

[image]

13) Launch *NxNandManager v4.0* and mount the drive agaaaain.

14) This time, restore the **'BCPKG2-1-Normal-Main.bin'** from the OFW eMMC file you saved to your computer and overwrite. 

15) Safely Eject!!

16) Done! Launch your emuMMC like you used to! (Probably also update your ipl.txts to new atm 0.13 / 10.x stuff.)