Atmosphere-NX/Atmosphere #1070: Atmosphere crash when JoyCon is removed from switch to play with bluetooth

DSMan97, 2020-07-04:
## Bug Report

When I am playing Zelda Breath of the Wild and I decided to remove the Joycons to play  separatly the Atmosphere crashed
 Atmosphere version 0.13.0
 Switch version 10.0.4
 Bootloader Hekate

SciresM, 2020-07-04:
Reopen using the issue templates.

closed 2020-07-04 by SciresM

DSMan97, 2020-07-04:
@SciresM Is already fixed for The Next version?

DSMan97, 2020-07-04:
@SciresM I usted The issue template, but have more information that I need yo share

masagrator, 2020-07-04:
> @SciresM I used The issue template, but have more information that I need to share

No, you don't.

DSMan97, 2020-07-04:
Okey I Will reopen later using the issue template and adding logs. Ithat help. Afer that I Will be forked the repo yo try to resolve by me own