Atmosphere-NX/Atmosphere #1072: Atmosphere 0.13.0 boots to black

onapjd, 2020-07-07:
## Bug Report

### What's the issue you encountered?

Updated atmosphere and firmware from 9.1.0 (and its atmosphere equivalent) to firmware 10.0.2 (did also attempt 10.0.4, but went to 10.0.2 when i read that it fixed this issue when someone else had it) and atmos 0.13.0. Now booting from latest Hekate or from official fusee causes a black screen after boot screen showing { 
R } </br>

### How can the issue be reproduced?

Download update for atmosphere and fusee. Install and replace any files it asks. Inject fusee (or hekate to boot fusee). Wait for splash screens. Black screen.</br>

### Crash Report

No crash reports made</br>

### System Firmware Version

10.0.2 and 10.0.4</br>

### Environment?

- Latest fusèe and hekate
- Official release
  - 0.13.0
 - Do you have additional kips or sysmodules you're loading: not that i am aware of</br>

### Additional context?

N/A

Shinyhunter2109, 2020-07-07:
Try to delete the old Atmosphere Folder from your SD Card and drag the new one to the SD Card . That could solve the Problem.

onapjd, 2020-07-07:
That seemed to have worked, no issues so far with booting applications or the hb menu. Thanks a ton.

closed 2020-07-07 by onapjd