Atmosphere-NX/Atmosphere #1077: Fatal error: Failed to initialize the SD card only on 256gb

monsterjj00, 2020-07-08:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

256gb sd card wont boot atmosphere.

### How can the issue be reproduced?

[ * ]</br>
Format 256gb sd card to fat 32
Drag and drop hekate and atmosphere
Launch hekate payload
launch fusee though hekate
</br>

### Crash Report

No crash reports were generated

### System Firmware Version

10.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - Hekate 5.3 (latest release)
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: N/A Clean install of Atmo

### Additional context?

My switch's sd card slot is okay due to my 128gb working just fine, and there are no errors on the 256 when testing through hekate and windows.

- Additional info about your environment:
- Nothing other than Hekate and atmosphere 13 are on the sd card and no files had been changed

masagrator, 2020-07-08:
Can you provide brand and model of sdcard?

monsterjj00, 2020-07-08:
> 
> 
> Can you provide brand and model of sdcard?

Its a PNY Pro Elite 256gb

SciresM, 2020-07-08:
"Won't boot" what logos do you see?

monsterjj00, 2020-07-08:
> 
> 
> "Won't boot" what logos do you see?

Nothing, no logos. Just a few seconds of black then fatal error.

SciresM, 2020-07-08:
"Fatal error" you mean you get an error screen?

monsterjj00, 2020-07-10:
> 
> 
> "Fatal error" you mean you get an error screen?

I suppose its an error screen, but its not a normal atmosphere blue screen like when it crashes while doing something.

Its just a black screen with the text rotated to be in portrait mode and it says "Fatal error: Failed to initialize the SD card"

cheesycod, 2020-07-12:
Try running atmosphere through fusee-primary instead if hekate

monsterjj00, 2020-07-12:
> 
> 
> Try running atmosphere through fusee-primary instead of hekate

Same thing

cheesycod, 2020-07-13:
The error you got means that your SD card was not recognised. Can you check hekate to see if it sees your card properly

monsterjj00, 2020-07-13:
> 
> 
> The error you got means that your SD card was not recognised. Can you check hekate to see if it sees your card properly and try booting via hekate fss0.

I dont exactly know what you mean by fss0. Though hekate sees it and runs fine but the second I enter atmosphere is just doesnt work.

cheesycod, 2020-07-13:
Do you have a line saying fss0=<path to fusee secondary>?

monsterjj00, 2020-07-13:
> 
> 
> Do you have a line saying fss0=<path to fusee secondary>?
If you mean add that line to the hekate ini Ill try that

monsterjj00, 2020-07-13:
> 
> 
> Do you have a line saying fss0=<path to fusee secondary> in hekates config or are you loading fusee-primary via hekate?

That didnt work either but I also don't know if I set it up right.

cheesycod, 2020-07-15:
can you post the contents of your hekate_ipl.ini

SciresM, 2020-07-15:
@cheesycod Why are you trying to debug for this guy on github issues?

The issue seems pretty clearly to be the SD card (hekate supports some sd cards that fusee doesn't).

I would try booting via hekate, yes, if this fails then horizon might fail on your SD card too, but hopefully it works. Either way, not much to be done.

closed 2020-07-15 by SciresM