Atmosphere-NX/Atmosphere #1079: Fatal error booting CFW

benbooth493, 2020-07-09:
## Bug Report

### What's the issue you encountered?

```shell
A fatal error occured when running Atmosphere.
Title ID: 010041544d530000
Error desc: std::abort() called (0xffe)
Report saved to /atmosphere/fatal_errors/report_0000000064rf54ea.bin

Press POWER to reboot
```

### How can the issue be reproduced?

Boot via latest `fusee-primary.bin`

### Crash Report

See attached.
[report_00000000064f54ea.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4895569/report_00000000064f54ea.bin.zip)

### System Firmware Version

10.0.3 (accidentally updated via Nintendo servers.)

### Environment?

N/A

### Additional context?

N/A

SciresM, 2020-07-09:
You installed non-exfat firmware but have an exfat sd card. Format your sd card as fat32, or manually flash exfat firm to package2.

closed 2020-07-09 by SciresM

benbooth493, 2020-07-09:
Ok, so are you saying despite my sdcard being exfat, the official firmware is fat32? Surely the official firmware installed via system update would be compatible?

SciresM, 2020-07-09:
That is what I'm saying.

And, nope. Presumably at some point you installed exFAT firm with choinx or something like it, and then if you did an official system update NS's savedata saw exFAT license not agreed to and installed FAT32 FIRM.