Atmosphere-NX/Atmosphere #1082: 0.13.0 Black Screen

Samekas123, 2020-07-10:
## Bug Report

Since version 0.13.0 I can't boot into the atmosphere, I get black screen. I have FAT32, tried also with fuseprimary and same thing happens black screen but this time with brightness. Going back to the version 12 everything works fine.

[ * ]</br>

### What's the issue you encountered?

Black screen initiating atmosphere. 

### How can the issue be reproduced?

[ * ]</br>
Inject payload henkate
Initiate sys
Black screen

### Crash Report

### System Firmware Version

10.0.4</br>

### Environment?

Hekate
Using official build
sdcreator
### Additional context?

It's only on this version I can't boot, on the others versions I can boot fine.

Samekas123, 2020-07-10:
Same problem being reported here: https://github.com/Atmosphere-NX/Atmosphere/issues/1053

Mindinfinity, 2020-07-11:
I had the same issue. After Formatting SD and making sure my Horizon Switch (Sysnand booted via normally, not RCM) was on latest firmware and my Emummc was on latest issue was resolved.

Samekas123, 2020-07-11:
> I had the same issue. After Formatting SD and making sure my Horizon Switch (Sysnand booted via normally, not RCM) was on latest firmware and my Emummc was on latest issue was resolved.

I have everything on latest version. I tried clean install many times.

Samekas123, 2020-07-11:
Looks like SciresM fixed this: https://github.com/Atmosphere-NX/Atmosphere/commit/1473adf5c4a6b7d833ceff60e8d8c3ed0fd02ab9 but I don't know how to compile it :/

cheesycod, 2020-07-13:
Firstly install devkitpro using https://switchbrew.org/wiki/Setting_up_Development_Environment, then follow docs/building.md

Samekas123, 2020-07-13:
> Firstly install devkitpro using https://switchbrew.org/wiki/Setting_up_Development_Environment, then follow docs/building.md

Thank you, I tried to compile it but getting some errors. Thanks anyways, was a good try! I will just wait for next build and hopefully this fixes will be merged.

cheesycod, 2020-07-13:
You have to use the latest master commit of libnx as well fyi

Feralzi, 2020-07-17:
Okay so idk why no one has said anything about this. Ill go straight to the point, theres a folder inside the nintendo folder on the root of your sd card, DELETE the private files. For some reason, after my switch updated, those two files popped up, and it wouldnt let me boot into my atmosphere (basically a black screen after the injection). Literally i spent 3 hours trying to find the source; had to use a second sd card that is formatted to check if my switch still runs atmosphere lol. Hope this helps.

marc-maurer, 2020-07-17:
@Feralzi For me its not solving the problem.

Samekas123, 2020-07-17:
> Okay so idk why no one has said anything about this. Ill go straight to the point, there's a folder inside the Nintendo folder on the root of your SD card, go inside the content folder, DELETE the "private" files. For some reason, after my switch updated, those two files popped up, and it wouldn't let me boot into my atmosphere (basically a black screen after the injection). Literally i spent 3 hours trying to find the source; had to use a second SD card that is formatted to check if my switch still runs atmosphere lol. Hope this helps.

Thank you for trying to help but still not working here.

> @Feralzi For me its not solving the problem. 
I went back to Atmosphère 0.12.0 and it is now working again.

Yep, going to version 12 solves it for me too.

shmimeow, 2020-07-20:
> @Feralzi For me its not solving the problem.
> 
> I went back to Atmosphère 0.12.0 and it is now working again.

#491

shmimeow, 2020-07-20:
> @Feralzi For me its not solving the problem.
> 
> I went back to Atmosphère 0.12.0 and it is now working again.

#metoo

Samekas123, 2020-07-21:
I don't know why still any dev replied to this thread.

punkgamer69, 2020-07-21:
> @Feralzi For me its not solving the problem.
> 
> I went back to Atmosphère 0.12.0 and it is now working again.

Same here. Going back to version 12 solves it for me.

SciresM, 2020-07-21:
So far as I know this is a duplicate of #1053, which I fixed a while ago. If it's not fixed in 0.14.0, re-open.

closed 2020-07-21 by SciresM