Atmosphere-NX/Atmosphere #1090: FS access log compiler breakage

DespairFactor, 2020-07-16:
When compiling AMS, I am hitting a compiler break, could be due to my compiler possibly, but wanted to bring this to your attention

/home/matthew/atmosphere/libraries/libstratosphere/source/fs/fs_access_log.cpp:192:22: error: '::fsOutputAccessLogToSdCard' has not been declared; did you mean 'OutputAccessLogToSdCardImpl'?
  192 |             return ::fsOutputAccessLogToSdCard(log, size);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                      OutputAccessLogToSdCardImpl
In file included from /home/matthew/atmosphere/libraries/config//../libvapours/include/vapours/results.hpp:23,
                 from /home/matthew/atmosphere/libraries/config//../libvapours/include/vapours.hpp:27,
                 from /home/matthew/atmosphere/libraries/libstratosphere/include/stratosphere.hpp:20:
/home/matthew/atmosphere/libraries/libstratosphere/source/fs/fs_access_log.cpp: In function 'void ams::fs::impl::{anonymous}::GetProgramIndexFortAccessLog(u32*, u32*)':
/home/matthew/atmosphere/libraries/libstratosphere/source/fs/fs_access_log.cpp:286:34: error: '::fsGetProgramIndexForAccessLog' has not been declared; did you mean 'GetProgramIndexFortAccessLog'?
  286 |                 R_ABORT_UNLESS(::fsGetProgramIndexForAccessLog(out_index, out_count));
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/matthew/atmosphere/libraries/config//../libvapours/include/vapours/results/results_common.hpp:253:39: note: in definition of macro 'R_ABORT_UNLESS'
  253 |         const auto _tmp_r_abort_rc = (res_expr); \
      |                                       ^~~~~~~~
/opt/devkitpro/devkitA64/base_rules:13: recipe for target 'fs_access_log.o' failed
make[3]: *** [fs_access_log.o] Error 1
Makefile:102: recipe for target 'lib/libstratosphere.a' failed
make[2]: *** [lib/libstratosphere.a] Error 2
make[2]: Leaving directory '/home/matthew/atmosphere/libraries/libstratosphere'
Makefile:8: recipe for target 'libstratosphere' failed
make[1]: *** [libstratosphere] Error 2
make[1]: Leaving directory '/home/matthew/atmosphere/libraries'
Makefile:36: recipe for target 'libraries' failed
make: *** [libraries] Error 2

Below is the commit that introduced this

https://github.com/Atmosphere-NX/Atmosphere/commit/e2b17086d45601c6dfc74f8af2274dcae130f724#diff-ec7c87c5ee65cebf43b09c224ea71d0b

HookedBehemoth, 2020-07-16:
Build and install libnx master.
`make install`

DespairFactor, 2020-07-17:
There is actually a problem with doing that, it will not install as the prereqs for building AMS includes libnx and libnx  is forced with other packages as a dependency

I got it working by installing only the dkp-linux packages and then compiling libnx master branch with the fix.

:: There are 9 members in group switch-dev:
:: Repository dkp-libs
   1) deko3d  2) libnx  3) switch-examples  4) switch-pkg-config
:: Repository dkp-linux
   5) devkit-env  6) devkitA64  7) general-tools  8) switch-tools  9) uam

closed 2020-07-17 by DespairFactor

fpb4, 2020-09-09:
sudo dkp-pacman -R switch-dev
checking dependencies...
Packages (10) deko3d-0.2.0-1  devkit-env-1.0.1-2  devkitA64-r16-1  devkitA64-gdb-9.2-1  general-tools-1.2.0-2  libnx-3.3.0-2  switch-examples-20200528-2  switch-pkg-config-0.28-4  switch-tools-1.8.0-1  uam-1.0.0-2

At the time of writing, libnx-3.3.0-2 is the latest release but make still complains.

~/dev/Atmosphere/libraries/libstratosphere/source/fs/fs_access_log.cpp: In function 'ams::Result ams::fs::impl::{anonymous}::OutputAccessLogToSdCardImpl(const char*, size_t)':
~/dev/Atmosphere/libraries/libstratosphere/source/fs/fs_access_log.cpp:192:22: error: '::OutputAccessLogToSdCardImpl' has not been declared; did you mean 'ams::fs::impl::{anonymous}::OutputAccessLogToSdCardImpl'?
  192 |             return ::OutputAccessLogToSdCardImpl(log, size);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                      ams::fs::impl::{anonymous}::OutputAccessLogToSdCardImpl
~/dev/Atmosphere/libraries/libstratosphere/source/fs/fs_access_log.cpp:190:16: note: 'ams::fs::impl::{anonymous}::OutputAccessLogToSdCardImpl' declared here
  190 |         Result OutputAccessLogToSdCardImpl(const char *log, size_t size) {
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~

Any clues pretty please?

HookedBehemoth, 2020-09-09:
Atmosphère builds of [libnx master](https://github.com/switchbrew/libnx).
`make install` that

fpb4, 2020-09-10:
Cheers @HookedBehemoth but no cigar.

Zap'ed it all, followed @DespairFactor 's advice above, hence went for dkp-linux only, together with the following documented dependencies: `switch-libjpeg-turbo, devkitARM and devkitarm-rules`  (https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/building.md)

```
:: Repository dkp-linux
   5) devkit-env  6) devkitA64  7) devkitA64-gdb  8) general-tools  9) switch-tools  10) uam
```

cloned, compiled and installed https://github.com/switchbrew/libnx master branch:
```
...skip...
bzip2 -cd libnx-3.3.0.tar.bz2 | tar -xf - -C /opt/devkitpro/libnx
make[1]: Leaving directory...skip...
```
```
make -j12
...skip...
fs_content.cpp
/home/moderator/dev/Atmosphere/libraries/libstratosphere/source/fs/fs_access_log.cpp: In function 'ams::Result ams::fs::impl::{anonymous}::OutputAccessLogToSdCardImpl(const char*, size_t)':
/home/moderator/dev/Atmosphere/libraries/libstratosphere/source/fs/fs_access_log.cpp:192:22: error: '::OutputAccessLogToSdCardImpl' has not been declared; did you mean 'ams::fs::impl::{anonymous}::OutputAccessLogToSdCardImpl'?
  192 |             return ::OutputAccessLogToSdCardImpl(log, size);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                      ams::fs::impl::{anonymous}::OutputAccessLogToSdCardImpl
/home/moderator/dev/Atmosphere/libraries/libstratosphere/source/fs/fs_access_log.cpp:190:16: note: 'ams::fs::impl::{anonymous}::OutputAccessLogToSdCardImpl' declared here
  190 |         Result OutputAccessLogToSdCardImpl(const char *log, size_t size) {
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
sc7fw_exception_vectors.s
...skip...
make[3]: *** [/opt/devkitpro/devkitA64/base_rules:14: fs_access_log.o] Error 1
```
Am i missing any further dependencies maybe?
```
~/dev/Atmosphere $ ls -lhtrA /opt/devkitpro/
total 28K
drwxr-xr-x 8 root      root      4.0K Sep  9 11:39 pacman
drwxr-xr-x 8 root      root      4.0K Sep 10 10:21 devkitA64
drwxr-xr-x 3 root      root      4.0K Sep 10 10:21 tools
drwxr-xr-x 5 root      root      4.0K Sep 10 10:21 licenses
drwxr-xr-x 3 root      root      4.0K Sep 10 10:21 portlibs
drwxr-xr-x 8 root      root      4.0K Sep 10 10:21 devkitARM
drwxr-xr-x 4 root      root      4.0K Sep 10 10:26 libnx
```

HookedBehemoth, 2020-09-10:
well you uninstalled deko3d by running this: `sudo dkp-pacman -R switch-dev`