Atmosphere-NX/Atmosphere #1093: Atmosphere 0.13 with 10.1.0

Surftown1, 2020-07-19:
I updates to fw 10.1.0 earlier and now i can't boot to cf tried formatting my sd card couple of times, didn't change anything, first i tried with hekate, now i'm trying with a pure atmosphere sd card.

I formated the sd card and downloaded the atmosphere 0.13 folder and copied the content to the sd card, then tried to boot using fusee-primary.bin.
and now i'm stuck in a atmosphere reboot loop

### Crash Report

[report_0000000005ac14b8.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4944661/report_0000000005ac14b8.zip)

### System Firmware Version

10.1.0

### Environment?

I am fairly certain that I'm running pure atmosphere right now.

hoping you guy's can help

HookedBehemoth, 2020-07-19:
exfat card and no exfat firmware

closed 2020-07-19 by SciresM

ELY3M, 2020-07-20:
please format your sd card with guiformat app then add your files back on.

Surftown1, 2020-07-20:
> exfat card and no exfat firmware

I'm sorry but this feels like a bad response, what firmware am I suppose to use then 10.0.4, 9.2?

I get this is no longer an Atmosphere problem, but how would i go about downgrading my firmware?

SciresM, 2020-07-20:
@Surftown1 ELY3M gave you advice.

The Atmosphere issue tracker is not a general purpose help forum; if it's not an atmosphere problem, you can seek help in some other switch user-support venue.