Atmosphere-NX/Atmosphere #1095: A fatal error occurred when running Atmosphere

ravrkr, 2020-07-23:
A fatal error occurred when running Atmosphére.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

Did you make any changes related to Atmosphère itself?
No

How can the issue be reproduced?
Boot to RCM
Launch Hekate
Boot Atmosphere

Crash Report
https://drive.google.com/file/d/1Fx7y_kN-0iAa0AKNFel4weoBL1UlGGDy/view?usp=sharing

System Firmware version
10.0.2

ravrkr, 2020-07-23:
please help me tried archive bits and extracting new atmosphere folder but still did not work

joer456, 2020-07-23:
Hi I am having the same issue. I believe this may be an issue with my SD card as I have been able to boot into AMS using a different card. 

I have tried using diskpart to fully clear down my SD card and load fresh AMS with no extras and I still get the same error.

I was on 10.0.4 and updated to 10.1.0 before the issue. I think the issue only came about when I started using the Hekate payload. I'll be sticking to the fusee primary payload now though.

Just wondering why, even after performing a disk clean and format, won't my original 128GB SD card work anymore?

[report_0000000005ab8872.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4964447/report_0000000005ab8872.zip)

SciresM, 2020-07-23:
@ravrkr your report is not publically accessible.

@joer456 You installed non-exfat firmware but have an exfat sd card. Format your sd card as fat32, or manually flash exfat firm to package2.

joer456, 2020-07-23:
@SciresM ah thank you! Appreciate the prompt response!

ravrkr, 2020-07-23:
@SciresM 
Heres another link
http://www.mediafire.com/file/0nbokl8xv0c2x42/report_00000000061b2685.bin/file

SciresM, 2020-07-23:
@ravrkr your issue really is archive bits. If the issue is persisting after running archive bit fixers (see hekate for how to do this), it may be your OS >.>

closed 2020-07-23 by SciresM