Atmosphere-NX/Atmosphere #1096: Atmosphere Crashes when going into Settings

Gr4ph1xZ, 2020-07-25:
## Bug Report

### What's the issue you encountered?

[ When I  go to Settings on my Switch it crashes and shows me this error, in airplane mode it works]
[image]
</br>
[ I use sys-ftpd and incognito ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ I use EmuMMC with sys-ftpd and incognito]</br>

### Crash Report

[ Crash reports was found in ``/atmosphere/fatal_reports``. ]</br>
[ <script src="https://gist.github.com/Gr4ph1xZ/2cec9cbe48804e7662b1739e8be4d7f8.js"></script>]</br>

### System Firmware Version

10.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: 
  - [ Official release version 0.13.0-master-32951329 ]
- Do you have additional kips or sysmodules you're loading: sys-ftpd and incognito
- Homebrew software installed: [ appstore, tinfoil,sigpatch-updater, atmosphere-updater ]

### Additional context?

- Additional info about your environment:
- [ N.A. ]

shchmue, 2020-07-25:
I think there was a theme problem with 10.1.10, did you use a custom theme? if so, manually delete it (`atmosphere/contents/0100000000001000`) and reinstall with the latest theme injector https://github.com/exelix11/SwitchThemeInjector/releases

Gr4ph1xZ, 2020-07-25:
> I think there was a theme problem with 10.1.10, did you use a custom theme? if so, manually delete it (`atmosphere/contents/0100000000001000`) and reinstall with the latest theme injector https://github.com/exelix11/SwitchThemeInjector/releases

I haven't installed a custom theme and the folder `atmosphere/contents/0100000000001000` is not existing on my sdcard.
i tried to reinstall the default theme with theme injector but it didn't work.

redondo56, 2020-07-25:
I had the same problem. Try putting the files somewhere safe and then format your SD card to ExFat and redownloading the newest version of Atmosphere. Test it with just the only the files of Atmosphere 0.13.0 and if it works, try passing all your files back to your SD card.

closed 2020-10-15 by SciresM