Atmosphere-NX/Atmosphere #1101: Fatal error when booting Atmosphere Error #010041544d530000

Roxann333, 2020-08-01:
Help please!

I updated my (banned) Nintendo Switch using Choidujournx to version 10.0.2 and my atmosphere (using SD Card set up) to the latest version as well. It worked perfectly until yesterday, I came home from work and the Switch had the error message showed in the attached file. I reformated my 256gb SD Card to EXFat using an IMac and added newly downloaded atmosphere files (exept the 'games' folder). Everytime add or update folders on my SD Card I run the folowing commands:
"sudo chflags -R arch /Volumes/nameOfMyCard/switch/"
"dot_clean /Volumes/nameOfMyCard"
"diskutil umountDisk /Volumes/nameOfMyCard"
Also, I fix the "Archive Bit" using Hekate and it usually works perfectly. I do have to re-install the games but it's a small price for me to pay and have the latest games and updates. I am unsure why it's not working now. Any help at all is very much appreciated!!!! 

[report_00000000055d4655.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5010971/report_00000000055d4655.bin.zip)

fincs, 2020-08-01:
> (banned) Nintendo Switch

> re-install the games

Piracy is not supported.

closed 2020-08-01 by fincs

Roxann333, 2020-08-01:
@fincs fuck you asshole. I was asking for help. I hope you fucking die.