Atmosphere-NX/Atmosphere #1102: Black screen upon booting Atmosphère 0.14.0

ThatNintendoNerd, 2020-08-01:
## Bug Report

### What's the issue you encountered?

Upon booting the latest release of Atmosphère, after the colored Atmosphère splash screen is displayed the console hangs on a black screen.

### How can the issue be reproduced?

Add contents of `atmosphere-0.14.0-master-a82914d5+hbl-2.3.4+hbmenu-3.3.0.zip` to the root of the microSD card and update `fusee_primary.bin`.

### Crash Report

N/A

### System Firmware Version

10.0.4

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
fusèe
- Official release or unofficial build: 
  - [ Official release version 0.14.0 ]
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: None

### Additional context?
- microSD card format is FAT32 formatted with guiformat with a 64 KiB allocation unit size.
- No other files on microSD card besides contents of Atmosphère 0.14.0 release ZIP.
- `fusee_primary.bin` has been updated with the 0.14.0 release.
- Reformatting the card and adding the Atmosphère 0.13.0 release contents to the root of the microSD card allows the console to boot fully.

SciresM, 2020-08-01:
Is this sysmmc or emummc?

Either way, please contact me on discord (SciresM#0524) for further support.

ThatNintendoNerd, 2020-08-01:
This is sysMMC. And I will do so.

Slluxx, 2020-08-02:
It works for me on 10.0.4 (same firmware as @ThatNintendoNerd) but i know one more who has problems with 0.14.0

muxi1, 2020-08-02:
The same problem under the same conditions applies to me under FW 9.2.0, but under a partition emuMMC.

spitfire, 2020-08-02:
same here, 1.0.1.0 sysMMC

matteopiccioni, 2020-08-02:
Same problem for me
sysmmc, sd fat32

THZoria, 2020-08-02:
same problem, test with 2 SD, one format in exFat (no other file)
and another SD format in fat32 (no other file) with also a
black screen via fusee primary / hekate in sysmmc

pontostroy, 2020-08-02:
Same for me with 10.1.0 sysMMC, exfat SD
Also I build master branch with the same result.

pixeltester, 2020-08-02:
Same issue on sysnand and firmware 9.2.0

SciresM, 2020-08-02:
Commenting "issue occurs" is largely unhelpful. A few people have volunteered to test; it'll be fixed either as soon as I can reproduce it or in the next few days.

maciak, 2020-08-02:
Hi.
I dont know if I have the same issue but for me it takes quite a while to boot. It gets stuck on color atmo logo for about a minute but then it boots fine. I'm on 10.0.0 btw

ImportTaste, 2020-08-02:
> Commenting "issue occurs" is largely unhelpful. A few people have volunteered to test; it'll be fixed either as soon as I can reproduce it or in the next few days.

Is there a way to generate some sort of debug log for you, then? There's no crash report.

NicholeMattera, 2020-08-02:
Not sure if this is related, but I was able to reproduce this issue by just adding an empty patch for loader. Just put the attached file (It's not a sigpatch it just contains "PATCHEOF") to sd:/atmosphere/kip_patches/loader_patches/, and it should produce a black screen after the Atmosphere splash screen.

[4D9C6D6993BF5BF058843BE072C9819598865C79BF87674C63B6D248FA7A0BA4.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5012307/4D9C6D6993BF5BF058843BE072C9819598865C79BF87674C63B6D248FA7A0BA4.zip)

ImportTaste, 2020-08-02:
It seems booting with hekate 5.3.2 works properly - doesn't produce the black screen. It's only with fusee-primary that things mess up.

fincs, 2020-08-02:
Fwiw, I have personally reproduced this issue with AMS 0.14.0 / fusee / sysver 10.1.0 without any loader patches (fake or not) being present.

ZeroGwafa, 2020-08-02:
> Fwiw, I have personally reproduced this issue with AMS 0.14.0 / fusee / sysver 10.1.0 (sysmmc) without any loader patches (fake or not) being present, i.e. only `default_nogc`.

Seconding this.  Had the same issue after updating.  Then wiped the SD card, and re-installed Atmosphere, changing only the BCT.ini file to disable nogc.  Black screen.  Removed that line from the BCT.ini file ,and it booted right up.

hexkyz, 2020-08-02:
Just to clarify, @ZeroGwafa are you now (after removing the nogc line) able to boot AMS fine every time?

Bappsack, 2020-08-02:
can also confirm the blackscreen on boot, tried with hekate and fg, fresh formated sd and default ams with no sys modules.

Latest FW, Fat32 Sysemmc

SciresM, 2020-08-02:
Managed to reproduce issue. Debugging with @hexkyz right now.

SciresM, 2020-08-02:
Figured out the issue. Fix shortly.

SciresM, 2020-08-02:
Hm, actually, only part of the issue was fixed by that. Soon enough.

commit 29358dc5 (2020-08-02): kernel_ldr: use unoptimized memcpy before mmu bringup (closes #1102)

Before the MMU is up, all reads/writes must be aligned; the optimized
memcpy implementation does not guarantee all reads/writes it performs
are aligned.

This commit splits the libc impl to be separate for kernel/kernel_ldr,
and so now only kernel will use the optimized impl. This is safe,
as the MMU is brought up before kernel begins executing.
closed 2020-08-02 by SciresM