Atmosphere-NX/Atmosphere #1103: Fatal when moving games from sd to system memory

annson24, 2020-08-02:
### What's the issue you encountered?

Having a fatal when transferring my games from microsd to system memory, doing the opposite does work tho. This is on emuMMC (vanilla atmosphere). I've already experienced this since AMS v0.13.0 but I was booting atmosphere via hekate back then because of the hex issue so I didn't bother reporting it before.

### How can the issue be reproduced?

1. Launch atmosphere via fusee-primary
2. Go to System Settings> Data Management> Move Data Between System/ microSD Card
3. Choose Move to System Memory
4. Choose any Game and select Move Data
5. Fatal occurs after a few seconds.

### Crash Report

https://gist.github.com/annson24/addb6fb1859b07dfc851fef5f47d2d88

### System Firmware Version

10.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  - fusèe
- Official release or unofficial build: 
  - Official release version 0.14.0
- Do you have additional kips or sysmodules you're loading:
  - no
- Homebrew software installed:
  - none

### Additional context?

- Additional info about your environment:
  - I'm booting fusee via pegascape and nereba. SysMMC is on v1.0.0.

SciresM, 2020-08-02:
Known issue with file-based emummc.

Should kinda be on the emummc repo and not the ams repo.

SciresM, 2020-08-24:
Closing this, because issue should be made on the emummc repo.

closed 2020-08-24 by SciresM