Atmosphere-NX/Atmosphere #1106: Cannot open any games after updating OFV to 10.1

issuesatmosphere, 2020-08-02:
Could not start the software.
Please try again from the HOME Menu.

##What's the issue you encountered?

My brother by mistake has updated my switch to the newest version of switch's firmware- 10.1, and because of that i can't enter any of games i've purchased. I tried to updating my atmosphere 0.13.0 to the newest one (0.14.0) but it didn't help.

##How can the issue be reproduced?

Anytime i try to start any game.

##System Firmware Version
10.1

##Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: 
  - Official release

- Homebrew software installed: Atmosphere 0.14.0

##Additional context?

If i used this issue template wrong i'm sorry, not really sure what to do with it, any tips would be welcomed. Thanks in advance for help

masagrator, 2020-08-02:
Cartridge games or eShop games?

issuesatmosphere, 2020-08-02:
eShop

Aliasim800, 2020-08-02:
Sigpatches?

issuesatmosphere, 2020-08-02:
i think they haven't been released for 0.14.0 yet, but i did have one for 0.13.0 version

fincs, 2020-08-02:
Piracy is not supported.

closed 2020-08-02 by fincs