Atmosphere-NX/Atmosphere #1107: Tinfoil not working

rhernandezr0930, 2020-08-03:
Hello, neither with 0.14.0 nor 0.14.1 of Atmosphere Tinfoil works, why in version 0.13.0 if it worked and in these not? It says that the program cannot be opened from home and closes.

[image]
[image]

fincs, 2020-08-03:
Piracy is not supported.

Also, you didn't use the issue templates.

closed 2020-08-03 by fincs

rhernandezr0930, 2020-08-03:
What templates are you referring to?

rhernandezr0930, 2020-08-03:
What templates are you referring to?  If it worked normal in the version of Atmosphère 0.13.0, and in this one in none of the ones they released it works.

Closed #1107<https://github.com/Atmosphere-NX/Atmosphere/issues/1107>.

rhernandezr0930, 2020-08-03:
What templates are you referring to? If it worked normal in the version of Atmosphère 0.13.0, and in this one in none of the ones they released it works.

Dr. Reynaldo Hernández Ruiz
Odontólogo
RUT 26.826.618-6

rhernandezr0930, 2020-08-03:
Can you answer me, please.

rhernandezr0930, 2020-08-03:
Thanks for nothing, for answers in this way, is that people prefer Xecuter.