Atmosphere-NX/Atmosphere #1109: Booting Issues

Ishidad, 2020-08-03:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

I can not boot, It crashes after showing the atmosphere logo. I updated tho the last atmosphere version  0.14.1 and after that I updated my fw through ChoiDujourNX to 10.1.0 and after that I encountered the problem.</br>

### How can the issue be reproduced?

[ * ]</br>
 I updated to atmosphere version  0.14.1 and after that I updated my fw through ChoiDujourNX to 10.1.0</br>

### Crash Report

https://gist.github.com/Ishidad/858b808467bf49884afff527ba2537c0</br>

### System Firmware Version

10.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate 5.3.2
- Official release or unofficial build: official release from github page
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: [ * ] gold leaf plus anything that came with atmosphere

HookedBehemoth, 2020-08-03:
Upload the report somewhere else or ZIP it and attach it here.
The gist is corrupted.

Ishidad, 2020-08-05:
Sorry I just seeing this, https://drive.google.com/file/d/1jGMYXnZU3ZBa1R7Ehp1Qm_fWWRDV4FZ5/view?usp=sharing it's on my gdrive

arch-box, 2020-08-05:
No one else has access to that gdrive file. Taking a stab in the dark, but did you have a custom qlaunch theme installed?

Ishidad, 2020-08-08:
You should have access now sorry for that, I just copy and paste the files, never added any custom theme

SciresM, 2020-08-08:
You installed non-exfat firmware but have an exfat sd card. Format your sd card as fat32, or manually flash exfat firm to package2.

closed 2020-08-08 by SciresM