Atmosphere-NX/Atmosphere #1115: Problem with /settings/data management

Rodry295, 2020-08-04:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

when i go into data management i crash the nintendo switch

### How can the issue be reproduced?

1.-I play in approximately 2 hours
2.-then I go into data management
3.-there he crashed me

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
https://gist.github.com/Rodry295/23fcce6c9a4202e3092117914fbb6ace

### System Firmware Version

10.1.0
### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version 0.14.1 ]
- Do you have additional kips or sysmodules you're loading:
no
- Homebrew software installed: [ * ]

no overlay but several homebrew apps like checkpoint, edizon, goldleaf, lithium, mmlinx,

### Additional context?

2162-0002
i used choi for the upds and i install hbg 3 days ago
but I think it has nothing to do with it because apart from uninstalling it, I already had this problem 1 week ago

SciresM, 2020-08-04:
pdm:qry -> QueryLastPlayTime() is being called, internal pdm object is returning count == 0 which leads to operation failure.

Your update/system saves are fucked.

Consider re-initializing console (deleting system savedata), then installing update with daybreak.

Not an atmosphere issue.

closed 2020-08-04 by SciresM

Rodry295, 2020-08-04:
What happens is that I already did another emunand

Rodry295, 2020-08-04:
where in the sd is the system savedata??

SciresM, 2020-08-04:
system savedata is not on the SD card. It is stored inside the emmc. You'll have to mount with e.g. HacDiskMount or equivalent, the Atmosphere issue tracker isn't the place for that kind of support though.