Atmosphere-NX/Atmosphere #1116: SYS-MODULES

EXO5KELETON, 2020-08-04:
Hi Fella,

Thank you for your hard work.

Any reasons why sys-modules seem to make AMS crash and not load games you get a black screen with the Nintendo symbol on the bottom right corner was ok with AMS 13.

Also exosphere.ini I used to use to blank my S/N if I use this in AMS14 on a clean install some of the games won't boot also the switch just freezes...the exosphere.ini should still remain on the root of the SD card am I correct? or has the location for this in AMS14 now changed?

Thanks a lot for your work.

SciresM, 2020-08-04:
Re-open using the issue templates.

This is your second issue opened without template usage in a row, third will lead to ban from atmosphere organization.

closed 2020-08-04 by SciresM

EXO5KELETON, 2020-08-04:
I can’t see where to do it.. when I select new issue it doesn’t let me do
anything else otherwise I would.

I don’t see how doing that will get me a ban.

SciresM, 2020-08-04:
When you make a new issue on github, you are made to select the issue template corresponding to your issue and fill it out. Fill out the issue on GitHub, if you must.

EXO5KELETON, 2020-08-04:
Ahhhh thank you so much.

Sorry for the confusion

EXO5KELETON, 2020-08-04:
Just filed another hopefully I have done it right this time

Thank you