Atmosphere-NX/Atmosphere #1117: SYS-MODULES

EXO5KELETON, 2020-08-04:
## Bug Report

### What's the issue you encountered?

[ When using sys-modules some games refuse to load and you get stuck with the Nintendo switch symbol bottom right corner and a black screen. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ No changes clean install of AMS14. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Install a system module like Sys-clk or Edizon Overlay. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[The Switch does not crash it just won't launch the games stays on the black loading screen with the Nintendo switch logo bottom right until you press the power button for 12/15 seconds for it to turn off ]</br>

### System Firmware Version

[Horizon Version 10.1.0 & AMS version 14.1]

### Environment?

[fusèe]

[Have Sys-Modules installed - Sys-Clk and Edizon Overlay]

### Additional context?

[N/A]

SciresM, 2020-08-04:
Probably you need to talk to the developers of those overlays/sys modules tbh. I can't provide support for them because I don't make them. I'm not aware of any changes to atmosphere that would impact those, unless they use ams libs and need a recompile or something.

Could be memory constraints, ams.mitm's FS/NCA pooled buffer logic uses a few extra MB. Either way, I would reach out to the developers of those sysmodules to get them to take a look.

EXO5KELETON, 2020-08-04:
Lovely thank you for your help.

Also am I correct that I still put exosphere.ini to the root of the SD card
to blank the s\n of my switch or does this now need to be in a different
location.

Thank you.

SciresM, 2020-08-04:
Correct, exosphere.ini's location and logic was not touched at all ^^

EXO5KELETON, 2020-08-04:
Perfect.

Thank you very much.

Enjoy your day and again thank you for your help and work with AMS.

👍🏽

SciresM, 2020-08-04:
No problem, hope you have a good day too. I'll close this for now, if you talk to the sysmodule developers and there turns out to be something wrong with ams re-open this issue ^^

closed 2020-08-04 by SciresM