Atmosphere-NX/Atmosphere #1119: Atmosphere does not wake from sleep mode

cadenmr, 2020-08-05:
## Bug Report

### What's the issue you encountered?

Atmosphere does not wake from sleep mode, requiring the power button to be held until system shuts down. This happens any time the system is put into sleep mode. Placing the switch in the dock does not produce any effect. 

### How can the issue be reproduced?

Put the system into sleep mode via power button, home menu or quick settings.

### Crash Report

None generated.

### System Firmware Version

10.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate 5.3.2 (launched via SX Core's payload menu)
- Official release or unofficial build: Official Release 0.14.1
- Do you have additional kips or sysmodules you're loading: Tesla, sys-clk, Status-Monitor, sys-tune
- Homebrew software installed: Daybreak, Retroarch, PPSSPP, NXEngine-evo, nxmtp, Reboot to payload

### Additional context?

- Additional info about your environment:
Switch is a patched Erista unit with a SX core installed.
exFAT 256GB micro SDXC card, formatted on the Switch.
Atmosphere is using a file-based emuMMC.
Both sysMMC and emuMMC are on the same version (10.1.0)

- What I've tried:
Removing all sysmodules (no effect)
Clean reinstall of Atmosphere (no effect)

SciresM, 2020-08-05:
@cadenmr Dump your bootrom patches with hekate and send them to me? 

I didn't know atmosphere even booted in that context. I can't claim it's really supported...

SciresM, 2020-08-05:
Yeah, reach out to me on discord (SciresM#0524). This is pretty much undebuggable otherwise.

cadenmr, 2020-08-05:
I've sent you a request.

SciresM, 2020-08-05:
Issue does not occur when chainloading fusee-primary.

closed 2020-08-05 by SciresM