Atmosphere-NX/Atmosphere #1120: Yet Another Fatal Error 0x4A2 (2162-0002) 0100000000001000

n3IVI0, 2020-08-06:
**What's the issue you encountered?**

Atmosphere crashes as soon as you get to the Home Screen, with the above error message.

**How can the issue be reproduced?**

Good question.

I have tried using 9.2.0, 10.0.4 and 10.1.0, and AMS 13.0, 14.0 and 14.1 --> same fatal error no matter what I do.

I finally tried reverting to my oldest backup, 5.1.0, and it remains stable. Anything newer --> KEERASH!!

**Crash Report**

https://gist.github.com/n3IVI0/e6c0e7175794bc02364841cf4badd91b

**System Firmware Version**

9.2.0, 10.0.4 and 10.1.0

**Environment?**

What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate 5.3.2

Official release or unofficial build: official release from github page

Do you have additional kips or sysmodules you're loading: only what came with Deepsea 1.5.1

However, I made a minimal version with SDSetup - no sysmodules, no Tesla, no overlays, only what was needed to get into homebrew --> same problem

With or without the sysmodules, I get the same crash.

I tried rolling back to 9.2.0 and running a minimalized AMS 13.0 --> same problem

I tried a full format (overwrite) of the SD card --> same problem

At no time has there been any custom themes, or NXTheme Installer. 

I tried keeping it in Airplane mode --> same problem

I have pored over this forum, and GBATemp. I tried all the advice I could find relevant to this crash. Nothing has worked for me. I am at a loss. My Switch is unusable in this state.

I really hope someone has a solution. I have attached the crash logs for both the 9.2.0 environment and the 10.1.0 environment.

Grass0001, 2020-08-06:
delete the folder sd/atmosphere/contents/01000000001000

n3IVI0, 2020-08-06:
That works, when that is the problem. That folder doesn't exist. There is no theme installed. There is no theme manager installed. I did an overwrite format to make absolutely sure. Nothing. It insists the folder is there - the folder is NOT there. I am baffled.

Masamune3210, 2020-08-06:
Make sure it doesn't exist in the titles folder either if it still exists

n3IVI0, 2020-08-06:
It's a clean install of Atmosphere. The new version uses /contents instead of /titles.

n3IVI0, 2020-08-06:
If I can make it into the homebrew menu before it crashes, then I'm good, but as soon as I go back to the Home Menu or Settings, within about five seconds, I get that fatal error.

I was able to turn off Tesla and verify all sysmodules are inactive, so I don't think it's that. In any case, running a version with no sysmodules has the same effect, gets the same crash.

n3IVI0, 2020-08-06:
It looks like uLaunch is still in pre-release, so probably too buggy to work as a replacement Home Menu. What could have borked my Home Menu, that persists in the face of different firmwares, different versions of Atmosphere, and when NO themes are installed?

n3IVI0, 2020-08-06:
https://github.com/WerWolv/Tesla-Menu/issues/21 Might be overlays and sysmodules conflicting with AMS 14, but I tried a stripped down version of AMS 14.1 and still get the crash. This is so weird.

n3IVI0, 2020-08-07:
I can update to 8.1.0. Everything works. No problems. If I go to 9.0.0 or higher, immediate crash. What changed from 8.1.0 to 9.0.0 that could be causing AMS to crash?

n3IVI0, 2020-08-07:
Tried running only Atmosphere, not hekate, loaded fusee-primary. Installed 10.1.0 with Daybreak. Still crashes. Tried the same thing on the SYSNAND, in case the problem was EMUMMC - still crashes. I cannot upgrade past 8.1.0 without it crashing on the Home Menu.

cheesycod, 2020-08-08:
Hmmm, can you try updating to 10.1.0 using ChoiDijourNX just to verify that this isnt a Daybreak issue

n3IVI0, 2020-08-08:
Done. Both Choi and Daybreak give the same result.

cheesycod, 2020-08-08:
Does your switch boot OFW? Have you tried using a different method of booting into CFW like using fusee-primary instead of hekate?

n3IVI0, 2020-08-08:
Yes. I used only Atmosphere. It gave the same result. I have been using Auto RCM since 6.2.0.

n3IVI0, 2020-08-08:
I have 7 burnt fuses, so that means I can go back to 6.0.1 right?

cheesycod, 2020-08-08:
Yes, although you said your issue began with 9.2.0 so it might not appear on 6.0.1.

n3IVI0, 2020-08-08:
Probably not. I stopped using OFW after I learned about burning fuses. I did make backups of every firmware I ran. The last one I had made was 9.1.0. When I restored the backup, which was working when I used it, it gave the fatal error. I can't figure what about Atmosphere could be causing this.

cheesycod, 2020-08-08:
> Yes. I used only Atmosphere. It gave the same result. I have been using Auto RCM since 6.2.0.

Then you have 8 fuses, not 7.

cheesycod, 2020-08-08:
Also, this should probably be in ReSwitched and not Github Issues

n3IVI0, 2020-08-08:
I checked. It's 7. I must have started using CFW just before 6.2.0 came out. If I remember right, that was the update you needed to be able to play Smash Brothers, and I got the banhammer right before it came out.

n3IVI0, 2020-08-08:
Whereabouts in ReSwitched?

cheesycod, 2020-08-08:
#user-support

n3IVI0, 2020-08-08:
Where can I get an invite?

n3IVI0, 2020-08-08:
Found it.

n3IVI0, 2020-08-08:
I can boot stock with zero issues. However, it is 6.01, not 9.00, so I am unsure what would happen at that level. I don't want to burn any fuses finding out. :)

cheesycod, 2020-08-08:
IDK why QLaunch is crashing for you

n3IVI0, 2020-08-08:
Somebody on ReSwitched suggested it could be an nvn crash (graphics driver).

n3IVI0, 2020-08-08:
I can get into maintenance mode in 9.0.0, but as soon as it is allowed to boot normally and I get into Home Menu - crash.

SciresM, 2020-08-08:
Crash occurs on stock using Nintendo's bootloader; seems to be hardware troubles (user mentioned graphical flickering in some game), and not an atmosphere bug.

closed 2020-08-08 by SciresM

drofdoom, 2020-08-08:
Seems to me like he deactivate the wifi auto connect and be fine but he
might of messed it up rolling back .he needs to update fully and call it a
day he should be good after that