Atmosphere-NX/Atmosphere #1121: Fatal when updating game

annson24, 2020-08-08:
### What's the issue you encountered?

On the latest vanilla atmosphere, my switch is throwing a fatal error whenever I try to update a game.

### How can the issue be reproduced?

1. Turn on the switch
2. Select the game (in my case it was overcooked! 2)
3. Select "update" when the console prompts us to update.
4. At around 12-13%, the screen will go black and a "A fatal error occurred when running Atmosphere" error appears.

### Crash Report

https://drive.google.com/file/d/1NPp-MOFvb8tVj4AKQlRLc4bRltpP7ufn/view?usp=sharing

### System Firmware Version

10.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  - fusèe
- Official release or unofficial build: 
  - Official release version 0.14.1
- Do you have additional kips or sysmodules you're loading:
  - none
- Homebrew software installed:
  - none

### Additional context?

I'm on file-based emummc.

SciresM, 2020-08-08:
The file you uploaded is an erpt.

A crash report or fatal report is needed to diagnose your error.

SciresM, 2020-08-08:
Ah, the file you uploaded actually does contain an error code -- fs/mmc port isn't active.

I'm gonna guess this, like your other issue, is an emummc (file-based) issue. I suspect there's some issue with file-based when file boundaries are crossed by operations.

Again, kind of an emummc repo issue, not an ams one.

SciresM, 2020-08-24:
Closing this, because issue should be made on the emummc repo.

closed 2020-08-24 by SciresM