Atmosphere-NX/Atmosphere #1124: Fatal error

Klepuna420, 2020-08-12:
#813 # What's the issue you encountered?
Fatal error.

### How can the issue be reproduced?
-Injekt Payload (hekate_ctcaer_5.3.2)
-Launch CFW (emuMMC)

### Crash Report
report_0000000008cf90de
Title ID 010000000000002b

### System Firmware Version
10.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
I use hakate

- Official release or unofficial build: 
Official release version hekate_ctcaer_5.3.2_Nyx_0.9.3

- Do you have additional kips or sysmodules you're loading:
no

- Homebrew software installed:
no

### Additional context?
no

SciresM, 2020-08-12:
You did not upload the crash report. Help cannot be provided until you upload it.

Klepuna420, 2020-08-12:
[report_0000000008cf90de.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5065231/report_0000000008cf90de.zip)
This should be the report. Sorry

kenman345, 2020-08-13:
im also having this issue. same error code. let me know if i can help provide information that helps figure this out as well

carcaschoi, 2020-08-13:
I have formatted sd card to fat32 and then problem solve

kenman345, 2020-08-13:
I have tried that without success
I've tried it multiple times, from scratch following different guides each time that all basically had me do the same things just some used sdsetup for setting my sd card up and others had me download the individual things they wanted me to use. I would really love for this to be figured out.
[report_08dfb229.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5069719/report_08dfb229.zip)

Klepuna420, 2020-08-13:
I also tried to format to fat32 but same error. I tried many different tutorials but always the same. My switch was hacked before this error. I formated my sd card and saved all the things on my desktop. Put them back on my sd card and this happened.

monaa9, 2020-08-15:
Ive been having the same issue. Formatted my sd card to fat32 and even tried the same thing with a different one but the problem persists.

kenman345, 2020-08-15:
So I still have this issue with my 512gb microSD card but I found a 64gb
microSD I had spare and set it up identically. Besides being unable to put
the Nintendo folder on it, i followed the exact same steps and it worked on
the first try and every time since then. But if I try my 512gb card it
simply gives me this error. I wonder if it has something to do with that.

kenman345, 2020-08-15:
When I get a chance I shall take a look and see if the extra pin is used
for the 512 that SDXC defines but is not usually on or needed for 64gb
cards.

Klepuna420, 2020-08-15:
Im using a 64 gb sd card and getting this error. I don't know if my sd card is the problem or any file is missing or something like this. Is there any tutorial which I should follow to be 100% sure that the problem is not me, doing anything wrong?

spencerdiniz, 2020-08-16:
I'm having the same issue. I tried 3 different SD cards (64GB and 128GB), with different formats (exFAT and FAT32).
This is happening on versions 0.14.0 and 0.14.1.

Eventually, I went back to 0.13.0, which is working fine.
I'll wait for another 0.14.x release before trying again.

For now, I'm running 0.13.0 on a 64GB exFAT micro SD without any issues.

kenman345, 2020-08-17:
hm...I'll try a downgrade of Atmosphere then too on my larger card since that would be more convenient

ther0n, 2020-08-19:
My SD card is formatted as exFAT and I was having a similar issue.

I fixed the issue by running Hekate then choosing the Tools tab. At the bottom tap on "Arch bit - RCM - Touch -Partitions", then click "Fix Archive Bit"

After doing that I was able to boot up successfully.

spencerdiniz, 2020-08-19:
> My SD card is formatted as exFAT and I was having a similar issue.
> 
> I fixed the issue by running Hekate then choosing the Tools tab. At the bottom tap on "Arch bit - RCM - Touch -Partitions", then click "Fix Archive Bit"
> 
> After doing that I was able to boot up successfully.

I tried "Fix Archive Bit", but it didn't work for.
Only downgrading to 0.13.0 worked.

SciresM, 2020-08-19:
fwiw @ "same issue" people, I really doubt it.

All errors in atmosphere show up to the user as "fatal error occurred", the particulars only really matter to me and are saved in the report binaries.

The issue the OP of this issue has is that his erpt save is somehow causing the save:/ mountpoint to be invalid when reading an attachment -- this is bizarre and I have literally no idea what would cause it, having triple checked the code since I received his reports. There's an obvious fix -- "delete the erpt save", which I would expect would work, but I haven't suggested it yet because I want to understand why their save causes the problem.

Everyone else *almost certainly* has different issues that you think are the same because of how the error display mechanism works but are really different. @spencerdiniz should create a new issue/upload his fatal error reports.

sdiniz-truelogic, 2020-08-19:
@SciresM, thanks for clarifying. 
I’ll reproduce the error again and create a new issue with the fatal error reports.
Will update here once I have the issue created.

spencerdiniz, 2020-08-19:
Can someone point me to documentation on how to retrieve the fatal error reports?
I was able to reproduce the fatal error, but I don't see any reports in the SD card (looked for *.bin files). 
Are the reports saved to Switch internal memory?

SciresM, 2020-08-19:
they're saved to sd:/atmosphere/fatal_errors -- the screen that displays on fatal error gives you the file path.

spencerdiniz, 2020-08-19:
@SciresM, thanks again.
My SD doesn't contain that directory. 
And the fatal error screen doesn't display a path. Instead, it prints out "Failed to save report to the SD card!".

SciresM, 2020-08-19:
Does the atmosphere dir itself have the archive bit set? If not, try creating a "fatal_errors" dir and make sure it doesn't have the archive bit set. This sounds like SD fuckery to me.

spencerdiniz, 2020-08-19:
> Does the atmosphere dir itself have the archive bit set? If not, try creating a "fatal_errors" dir and make sure it doesn't have the archive bit set. This sounds like SD fuckery to me.

@SciresM, first of all, thanks for your attention and time.
There's definitely some fuckery... 

I created the "fatal_errors" directory manually, reproduced the error and crash report was successfully saved.
But, when I tried to read the SD card in Windows to retrieve the report, I got a message that the SD was not formatted, and was not able to retrieve the report.

So, I started the whole thing again (determined to get the report). I formatted the SD to exFAT, but this time, I manually created the "fatal_errors" directory from the get-go (since I knew I would need it). However, when I tried to reproduce the error, everything worked and Atmosphere booted.

So I tried a 3rd time (with the original steps). Formatted, copied Atmosphere to SD and injected fusee-primary.bin... and... No fatal error!

I can no longer reproduce the fatal error.
I going to try again... and update here if I can manage to reproduce again.

Best regards,
Spencer

Klepuna420, 2020-08-19:
I downgraded now to 13.0.0 and it works. Its something with the new atmosphere version I think. I think I'm just waiting for a new atmosphere version. I also tried to "Fix Archive Bit" But still got this crash. Is there anything else that I should try?

LunaerD, 2020-08-28:
Is there any new info about this fatal error ? i just updated Hekate and atmosphere to the latest release but i'm experiencing the exact same crash.

spencerdiniz, 2020-08-28:
I was not longer able to reproduce this error with a difference SD card. 
My previous SD card seemed to be broken.

allanleone, 2020-09-10:
Hi everyone. 

I'm facing the this problem since installed atmosphere 14 (but unable to solve using any old versions from now on). 

My device it's stucked into the firmware 5-8 (5 restoring via choidjour for windows, < 8 directly via nx choidjour)

any other firmware (9, 10+) that I try to install, the system does not boot. I test it multiple SD cards, sysNAND / emuMMC, and the fatal error persists. any idea or suggestions of how to proceed? thank you all!

[logs.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5202667/logs.zip)
[report_0113bc9d0.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5202675/report_0113bc9d0.bin.zip)

Klepuna420, 2020-09-12:
Hello,

I tried to install the new 14.2 version but still got the same error.
Still have to use atmosphere 13.0...
Here is the report if it can help in any way:

[report_000000000911b1ee.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5213233/report_000000000911b1ee.zip)

iMaxed0ut, 2020-09-20:
Same issue when updating to 0.14.4 and Hekate 5.3.3. Panic error causing a crash in the boot process. 

These are the bin reports from the times its crashed. 
[Bin Reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5252260/Bin.Reports.zip)

it was on folder 0100000000000042. I found that deleting the directory it was referencing allowed me to successfully boot into CFW.

Klepuna420, 2020-09-23:
I fixed the error by deleting the folder 010000000000002b in Atmosphere/Contents/ to solve this problem. Hope it did help you too.

closed 2020-09-24 by SciresM

KamiSquadX, 2022-04-26:
has this issue been resolved yet??

takoyaki99, 2022-04-26:
> has this issue been resolved yet??

> @nwhiteman42 Please don't try to provide support for issues you don't understand.
> 
> @Leonelvader This is actually the same issue as #1124, #1145, #1173 and was fixed recently. The next Atmosphere release won't trigger the problem.
> 
> Zip containing fixed erpt-sysmodule (extract to your SD card, replacing the 0.14.4 release one) [here](https://github.com/Atmosphere-NX/Atmosphere/files/5273156/atmosphere-0.14.4-erpt-fix.zip).

KamiSquadX, 2022-04-26:
thanks for the swift reply. so i downloaded the atmosphere-0.14.4-erpt-fix so this drag and drop should fix it? trying it now

takoyaki99, 2022-04-26:
> thanks for the swift reply. so i downloaded the atmosphere-0.14.4-erpt-fix so this drag and drop should fix it? trying it now

Considering how said issue has been closed, I think using the latest Atmosphere will completely fix this for you. I quoted Scires to say "yes, it's been fixed."
Apparently if the issue still persists, it could be your SD or reader being messed up, according to some comment I found.

Also, this is an old issue--please don't comment on those! If you want support go to the ReSwitched Discord, if you've found a bug, please open up a new report~

KamiSquadX, 2022-04-26:
I got this error from using the latest atmosphere. never had this happen before. it happened after repeatedly trying to load retroarch etc, atmosphere crashes, restart using payload injector dongle, repeat all this like 3 times, until trying to launch cfw gives the black screen fatal error message. Where is the discord link cause i still need help with this.
[report_0000000001566de7.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8559887/report_0000000001566de7.zip)

takoyaki99, 2022-04-26:
> I got this error from using the latest atmosphere. never had this happen before. it happened after repeatedly trying to load retroarch etc, atmosphere crashes, restart using payload injector dongle, [...]

https://discord.gg/ZdqEhed good luck!!

commit 4138abbe (2020-09-24): erpt: fix attachment-in-save paths (closes #1124, #1145)