Atmosphere-NX/Atmosphere #1126: Bug Report

hungpham1309, 2020-08-15:
A fatal error occurred when running Atmosphere.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)
Report saved to /atmosphere/fatal_errors/report_00000000066a9486.bin

What's the issue you encountered?
After booting up Atmosphere, through Hekate, screen goes black and I receive the fatal error message

How can the issue be reproduced?
Boot to RCM
Launch Hekate
Boot Atmosphere through either fusèe-primary or emunand

Crash Report

https://github.com/hungpham1309/bug/blob/master/report_00000000066a9486.bin

System Firmware Version
9.2.0

HookedBehemoth, 2020-08-15:
exFAT card without having exFAT firmware installed.
Format your sd card to FAT32.

hungpham1309, 2020-08-15:
> exFAT card without having exFAT firmware installed.
> Format your sd card to FAT32.

Thanks, the issue is resolved now.

closed 2020-08-15 by hungpham1309