Atmosphere-NX/Atmosphere #1127: Question / Pipe dream, running Atmosphère on Nvidia Shield?

sigboe, 2020-08-15:
## Feature Request

### What feature are you suggesting?
#### Overview:
AFAIK Atmosphere aims to replace the whole Horizon firmware, bit by bit, am I right?
If so, after all or most of the code has been re-implemented. Do you have any aspirations of running Atmosphère on other Tegra X1 systems, such as Nvidia Shield? I understand this might be tagged as wontfix, even without committing to any future action. But at the very least do you have any options or insight regarding the topic of running Atmosphère on non switch devices?

#### Smaller Details: 
N/A

#### Nature of Request:  

Porting Atmosphère to other Tegra X1 systems?

### What component do you feel this would best fit within?

All

### Why would this feature be useful?
Would be cool to run Switch software on more powerful hardware, with better power delivery, higher by default.
Maybe so people could research Horizon/Atmosphère without having native hardware. 
Even if you could not run Switch software on another platform it would be a cool curiosity to run Horizon/Atmosphère on another platform. I believe I herd in a stream you think Horizon is an awesome operating system under the hood.
_Even if this is plausible it could take a very long time, longer than the lifetime of the current Switch, but I am amused by the though of trying out AI upscaling on the 2019 Nvidia Switch in Atmosphère before a new Switch is released with this built in._

masagrator, 2020-08-15:
AI upscaling used in 2019 Shield is a fully software implementation taxing hardware. You would need to implement similar resolution to Horizon first, which means it could run even on native Switch hardware but in comprosing games performance.
And if new Switch will still use Nvidia hardware, new SoC will probably contain Tensor cores and thus using DLSS which will be way better than what is Shield using now.

My few words before this issue will be closed. 👍

fincs, 2020-08-15:
Friendly reminder that every single Nintendo Switch game ships with a baked-in GPU driver specific for the console with no clear way to swap it out for something else, and also tend to ship with shaders compiled ahead of time specifically for the Maxwell ISA variant used on the Tegra X1. Changing to a non-Maxwell GPU is not possible without every single developer literally recompiling every single game released on the Switch for the new platform.

SciresM, 2020-08-15:
Gonna close this because an issue isn't really the appropriate place for it.

That said, I'm very interested in exploring Horizon/Atmosphere on other boards. I intend to port mesosphere to x64 after BootImagePackage reimplementation concludes (targeting by end of next yearish).

That said, I am not aiming to e.g. play commercial software on other boards -- even besides the obvious anti-piracy reasons, I have roughly negative interest in writing a gpu driver/nvservices equivalent at the present time.

closed 2020-08-15 by SciresM